nanoframework/Home #962: GetCustomAttributes for Properties

KlausVcb, 2022-02-07:
### Details about Problem

ESP_WROVER_KIT

Firmware build Info:
  Date:        Jan 14 2022
  Type:        MinSizeRel build, chip rev. >= 0, support for PSRAM
  CLR Version: 1.7.3.225
  Compiler:    GNU ARM GCC v8.4.0

### Description

There is no Type.GetProperties support yet. Instead the getter and setter methods are returned when executing Type.GetMethods. Executing GetCustomAttributes on these MethodInfo's does not return custom attributes.

### Detailed repro steps so we can see the same problem

1. Execute the following program:

[CustomAttributes.zip](https://github.com/nanoframework/Home/files/8014418/CustomAttributes.zip)

2. This will output the following:

Found 0 attributes on get_MyProperty, expected: 1
Found 0 attributes on set_MyProperty, expected: 1

### Expected behaviour

Two solutions can be envisioned:
- Returning custom attributes on properties with the getter and setter methods
- Implementing Type.GetProperties and PropertyInfo.GetCustomAttributes

Ellerbach, 2022-02-08:
I've been using for the WebServer custom attributes on methods. And here is a pattern you can use:
https://github.com/nanoframework/nanoFramework.WebServer/blob/bed89ad5d22adbfb21510457ae2042a0dbb5cce1/nanoFramework.WebServer/WebServer.cs#L185
What is missing from this pattern? I didn't feel I was blocked in any way.

KlausVcb, 2022-02-09:
It does work on regular methods. It does not seem to work on getter_ and setter_ methods of properties.
That's the only way I could see to get at attributes on properties since there is no Type.GetProperties().
The [PropertyInfo](https://docs.nanoframework.net/api/System.Reflection.PropertyInfo.html) class does exist however.

In short, the issue boils down to being unable to get the PropertyInfo instances of a type. GetCustomAttributes() on methods relates to the getter_ setter_ methods of properties which was an attempt at a workaround.

Ellerbach, 2022-02-09:
> It does work on regular methods. It does not seem to work on getter_ and setter_ methods of properties.
> That's the only way I could see to get at attributes on properties since there is no Type.GetProperties().

Well, you can do that as easy as this: https://github.com/nanoframework/nanoFramework.Json/blob/0052f7359832682bcd823a449d646c1adfde9174/nanoFramework.Json/JsonConvert.cs#L301

> In short, the issue boils down to being unable to get the PropertyInfo instances of a type. GetCustomAttributes() on methods only relates to the getter_ setter_ methods of properties which was an attempt at a workaround.

All setter methods starts with `set_`, all getter methods starts with `get_`

So nothing is blocking you at all.

KlausVcb, 2022-02-09:
Fair enough except that executing GetCustomAttributes on the MethodInfo you get from the get_ or set_ method returns empty. See the attached solution for an easy repro.

closed 2022-04-28 by stale[bot]