nanoframework/Home #965: Nonunique keys for Hashtable when using value types

alexxxz1977, 2022-02-11:
### Target name(s)

 ESP32_PSRAM_REV0-1.7.4

### Firmware version

1.7.4.285

### Was working before? On which version?

1.7.1.1090

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.3.2
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Feb  2 2022
  Type:        MinSizeRel build, chip rev. >= 0, support for PSRAM
  CLR Version: 1.7.4.285
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  Softprise.NanoCRT.TestApp, 1.0.0.0
  System.Device.Spi, 1.0.3.0
  System.Math, 1.4.4.0
  mscorlib, 1.12.0.0
  System.Device.Gpio, 1.0.3.0
  nanoFramework.System.Collections, 1.4.0.0
  nanoFramework.Runtime.Native, 1.5.4.0
  nanoFramework.Runtime.Events, 1.10.0.0

Native Assemblies:
  mscorlib v100.5.0.16, checksum 0xE7505F22
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x210D05B1
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.0, checksum 0x5B533478
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
[... 243 lines cut ...]
>   Region     Start      Blocks   Bytes/Block    Usage
> +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
>       0    0x00010000       1      0x1A0000     nanoCLR
>       1    0x001B0000       1      0x1F0000     Deployment
>       2    0x003C0000       1      0x040000     Configuration
> 
> 
> +++++++++++++++++++++++++++++++++++++++++++++++++++
> ++              Storage Usage Map                ++
> +++++++++++++++++++++++++++++++++++++++++++++++++++
>   Start        Size (kB)           Usage
> +++++++++++++++++++++++++++++++++++++++++++++++++++
>   0x003C0000    0x040000 (256kB)    Configuration
>   0x00010000    0x1A0000 (1664kB)   nanoCLR
>   0x001B0000    0x1F0000 (1984kB)   Deployment
> 
> 
> 
> Deployment Map
> Empty
> 

### Aditional information

I don't think that the issue is ESP32 specific. I think you can have the same behavior on other platforms but I don't have another hardware to check this.

josesimoes, 2022-02-11:
@alexxxz1977 this is happening because of the recent changes in the issue templates and adjustments in nfbot.
The bot is still adapting to this new reality... 😉 

I'll reopen it in a few minutes as the fix is going through the pipeline.

josesimoes, 2022-02-11:
@alexxxz1977  thank you very much for the detailed issue report! 👍🏻

alexxxz1977, 2022-03-02:
@josesimoes 

> @alexxxz1977 with this you may have uncovered another potential issue in the execution engine... Running it on full .NET the code above fails because one can't change a collection inside a loop that's used on the iterator. Test app [here](https://dotnetfiddle.net/i7G2ZY).

Why the code above works fine on 1.7.1.1090? I had to use this FW because doesn't have any issues. I thought that 1.7.1.1090 and 1.7.4.285 have the same C# version.

josesimoes, 2022-03-03:
@alexxxz1977 these differences have nothing to do with the C# language version, rather with the native implementation that supports the `GetHashCode()`.

closed 2022-03-03 by josesimoes