nanoframework/Home #967: System.Device.Gpio.PinValuePair resolves to unknown type

mork2020, 2022-02-15:
### Library/API/IoT binding

System.Device.Gpio

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.1.0.21

### Target name(s)

ESP32_REV0

### Firmware version

1.7.4-preview.36

### Device capabilities

** .NET nanoFramework extension v2022.1.0.21 loaded **
GitHub repo: https://github.com/nanoframework/Home
Report issues: https://github.com/nanoframework/Home/issues
Browse samples: https://github.com/nanoframework/samples
Join our Discord community: https://discord.gg/gCyBu8T
Join our Hackster.io platform: https://www.hackster.io/nanoframework
Follow us on Twitter: https://twitter.com/nanoframework
Follow our YouTube channel: https://www.youtube.com/c/nanoFramework
Star our GitHub repos: https://github.com/nanoframework/Home
Add a short review or rate the VS extension: https://marketplace.visualstudio.com/items?itemName=nanoframework.nanoFramework-VS2022-Extension

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Feb 11 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.7.4.36
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  NFApp1, 1.0.0.0
  System.Device.Gpio, 1.0.3.0
  nanoFramework.Runtime.Events, 1.10.0.0
  mscorlib, 1.12.0.0

Native Assemblies:
  mscorlib v100.5.0.16, checksum 0xE7505F22
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x210D05B1
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.0, checksum 0x5B533478
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.3.0, checksum 0xA67E6317
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.0.0, checksum 0x48031DC5
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

System.Device.Gpio.PinValuePair can't be resolved while starting the program on the target.

### How to reproduce

1. Deloy the my demo project to the ESP32 target

### Expected behaviour

System.Device.Gpio.PinValuePair can be used

### Sample project or code

https://github.com/mork2020/ErrorPinValuePair

### Aditional information

The package folder content is:

nanoFramework.System.Device.Gpio.1.0.3-preview.8
nanoFramework.CoreLibrary.1.12.0-preview.5
nanoFramework.Runtime.Events.1.10.0-preview.6

mork2020, 2022-02-24:
Update - after MCP23xxx is availbale as a device in nanoframework IoT (Iot.Device.Mcp23xxx, 0.0.0.0).
Using this device fails in my ESP32 project with the same error:

The program '[1] .NET nanoFramework application: Managed' has exited with code 0 (0x0).
Resolving.
Resolve: unknown type: System.Device.Gpio.PinValuePair
Error: ff000000
Waiting for debug commands...

--> Is there any workaround of this problem?

##############################################################################

Here my device capabilities:

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Feb 24 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.7.4.63
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  NFApp1, 1.0.0.0
  System.IO.FileSystem, 1.0.3.0
  nanoFramework.System.Text, 1.1.3.0
  nanoFramework.ResourceManager, 1.1.4.0
  nanoFramework.Hardware.Esp32, 1.3.5.0
  System.IO.Streams, 1.0.0.0
  nanoFramework.Runtime.Events, 1.10.0.0
  System.Device.I2c, 1.0.3.0
  UnitsNet.Temperature, 4.121.0.0
  nanoFramework.Runtime.Native, 1.5.4.0
  System.Device.Gpio, 1.0.3.0
  Iot.Device.Mcp23xxx, 0.0.0.0
  mscorlib, 1.12.0.0
  nanoFramework.System.Collections, 1.4.0.0
  Iot.Device.Mcp960x, 1.0.0.0
  UnitsNet.Length, 4.121.0.0

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.1, checksum 0xA01012C3
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.4.0, checksum 0xCB7C0ECA
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.0.0, checksum 0x48031DC5
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

##############################################################################

Ellerbach, 2022-02-25:
> Resolve: unknown type: System.Device.Gpio.PinValuePair

Hum, that is weird. If a type doesn't exist, the project can't build.
Sorry for asking as on my phone: is it the very last version of the nuget you are using?
Also seems weird about the version.

mork2020, 2022-02-28:
Everthing seems up to date. I reduced the example to create only one instance of System.Device.Gpio.PinValuePair and cant get it working. Actually i can't figure out, what i'm doing wrong.

josesimoes, 2022-03-01:
Reproduced. Fix incoming.

closed 2022-03-02 by josesimoes