nanoframework/Home #968: Assembly.Load(buff) when buff is loaded from internal storage fails if done after the read

Ellerbach, 2022-02-16:
### Library/API/IoT binding

System.Reflection.Assembly

### Visual Studio version

VS2022

### .NET nanoFramework extension version

2022.1.0.21

### Target name(s)

ESP32 REV 0 with and without PSRAM

### Firmware version

1.7.4-preview.33

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Feb 14 2022
  Type:        MinSizeRel build, chip rev. >= 0, support for PSRAM
  CLR Version: 1.7.4.33
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.16, checksum 0xE7505F22
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x210D05B1
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.0, checksum 0x5B533478
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.3.0, checksum 0xA67E6317
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.0.0, checksum 0x48031DC5
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3f80086c  0x003b0000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

When loading an assembly stored in the internal memory I:, the Assembly.Load below fails if no Thread.Sleep is placed:

```csharp
            using FileStream fspe = new FileStream(file, FileMode.Open, FileAccess.Read);
            Debug.WriteLine($"{file}: {fspe.Length}");
            var buff = new byte[fspe.Length];
            fspe.Read(buff, 0, buff.Length);
            var ass = Assembly.Load(buff);
            // This creates an exception
```

Exception created:
```text
    ++++ Exception System.Exception - CLR_E_FAIL (1) ++++
    ++++ Message: 
    ++++ System.Reflection.Assembly::Load [IP: 0000] ++++
    ++++ Program+<>c__DisplayClass0_0::<<Main>$>g__LoadAssemblies|2 [IP: 0072] ++++
    ++++ Program::<Main>$ [IP: 003f] ++++
```

This actually works:

```csharp
            using FileStream fspe = new FileStream(file, FileMode.Open, FileAccess.Read);
            Debug.WriteLine($"{file}: {fspe.Length}");
            var buff = new byte[fspe.Length];
            fspe.Read(buff, 0, buff.Length);
            Thread.Sleep(1);
            var ass = Assembly.Load(buff);
```

### How to reproduce

1. Create code to save an assembly in the I: storage
2. use the code up

### Expected behaviour

The Thread.Sleep should not be necessary and the Load should just work after the read

Ellerbach, 2022-04-19:
This is still valid dear bot :-)

networkfusion, 2024-12-16:
@Ellerbach is this still valid?

Ellerbach, 2026-09-28:
@Wim-Dekker are your pe file in Release or in Debug? They should be in Release!

Wim-Dekker, 2026-09-29:
@Ellerbach I've tested with both debug and release versions. Both failed most of the time. Currently I'm relying on a failure detection system that automatically reboots the controller until all assemblies are loaded correctly. This can take several hours.

closed 2026-10-30 by josesimoes