nanoframework/Home #969: Update esptool in nanoff

josesimoes, 2022-02-16:
### Details about Problem

**nanoFramework area:** ( nanoff)

### Description
Need to rebuild esptool in all platforms as it was updated in the latest IDF 4.4.

josesimoes, 2022-02-17:
I've improved the process so the local packaging with the Python tool in every platform is not required anymore.
It now downloads the binaries from the latest release on GitHub and places them on the appropriate tool folders.

closed 2022-02-17 by josesimoes

Ellerbach, 2022-02-17:
> I've improved the process so the local packaging with the Python tool in every platform is not required anymore. It now downloads the binaries from the latest release on GitHub and places them on the appropriate tool folders.

Very very very nice improvement! One less pain to update :-)