nanoframework/Home #970: Add Support for ESP32 S3

jonstelly, 2022-02-16:
### Description

I got a new devkitc ([link](https://www.amazon.com/gp/product/B09D3S7T3M/)) and got an error trying to flash.  I had some brief discord discussion with Jose and see he created #969 to update esptool in nanoff but it looks like S3 might also need new targets for this board/chip?

When I dropped esptool 3.2 into nanoff I got the following error trying to flash the above board:

`[Errno 2] No such file or directory: '...\\nanoff\\2.0.15\\tools\\net6.0\\any\\esp32s3bootloader\\bootloader.bin'`

### How to solve the problem

I'm unsure of any of this but from trying to build the firmware using the documentation

- it seems like maybe a new cmake/gcc setup (`xtensa-esp32s3-elf.cmake`?)
- New docker images including above?
- New ESP32_S3 target (clone of ESP32_S2)?

josesimoes, 2022-02-16:
Yes support for S3 is still on the works. It was taken into consideration during the last rework of the build system.
Curently waiting to get my hands on one of those dev kits to get it rolling. Please be patient. 😁

jonstelly, 2022-02-16:
Just sponsored the project for $25.  It's not in any way donated only for this so feel absolutely free to spend it on something other than the S3 devkit.  But if you do order one I didn't want you to have to pay for it.  My office is a mess of various models of devkits and I've only been focusing on ESP32, so I can only imagine yours is worse.  Thanks for all your work.

josesimoes, 2022-02-16:
I try to tidy it every now and then... 😜

On behalf of the project that you very much for the donation and your kind words. Those mean a lot!

babyblue2, 2022-05-25:
I'm also looking forward to nanoff for ESP32_S3

AachoLoya, 2022-07-04:
What's the expected timeline for S3 support?

bghavami, 2022-07-15:
Arduino has full support for S3 now. I am patiently waiting to see the same happening to nanoframework.

Ellerbach, 2022-07-16:
> Arduino has full support for S3 now. I am patiently waiting to see the same happening to nanoframework.

A lot of people are. But please keep in mind .NET nanoFramework is a community effort except for what is sponsored. There are people spending their nights and weekends adding features, adding board support, adding classes. So please don't be too impatient. Also, we do love PR and contributions. So with a bit of everyone help, we can all achieve a lot of great things!

peirens-bart, 2023-01-12:
I have a devkit for the S3, can I help to make or test the build if exists?

josesimoes, 2023-01-12:
> I have a devkit for the S3, can I help to make or test the build if exists?

Thanks for offering! If you're familiar with C/C++ and Espressif IDF you sure can help.
As for the build, there is none at this time.

peirens-bart, 2023-01-12:
> > I have a devkit for the S3, can I help to make or test the build if exists?
> 
> Thanks for offering! If you're familiar with C/C++ and Espressif IDF you sure can help. As for the build, there is none at this time.

I have in both some, I never contributed to an open source project before so probably need some coaching for that as well.
Is there a procedure wiki for creating a new build some where?

josesimoes, 2023-01-12:
@peirens-bart there are docs on how to setup the build and work with the dev containers. On how to make an ESP32 port, there are none. Reason is that it's a very very specific (and broad) matter and it's only required when Espressif releases a new family. Therefore it's not worth the effort on writing and maintaining such document. (that time would be much more well spent on actualy doing the port 😉 )

Know that there is another community member that has offered to tackle this.
Suggest that you head over to our Discord server and touch base with him. Here's the link for the thread:
https://discord.com/channels/478725473862549535/481782754674212867/1061015039940776037

2505817596, 2023-03-01:
What's the expected timeline for S3 support?

josesimoes, 2023-03-01:
Right now this is on hold. There aren't enough resources to invest on this.

It seems there is a lot of interest on having support for this ESP32 series. 
FYI [greenyng](https://twitter.com/greenyng_) it's willing to co-sponsor this. Recently M5Stack has showed interest in co-sponsoring it too, see this [tweet](https://twitter.com/M5Stack/status/1625860079491887104). But this has gone nowhere.

JORGEGO, 2023-03-01:
Clear!!! We could join forces and sponsor togethers, what do you thin @2505817596?

JORGEGO, 2023-03-02:
Dear all, we need to confirm the sponsor cost to ESP32-S3FN8 (M5Stack in my case). Base on budget and people interested to sponsor from greenYng we could invest some quantity, so how many people could invest and what it is the total amount to develop the firmware ESP32-S3FN8? we could crowdfunding this project, please add people interested and amount to invest and @josesimoes confirm you minimal budget to develop it, thanks

yltsa, 2023-03-14:
Maybe Unexpected Maker has interest to sponsor for a support of their new ESP32-S3 based boards..?? Or LilyGo..

mzirkle44060, 2023-04-16:
> Dear all, we need to confirm the sponsor cost to ESP32-S3FN8 (M5Stack in my case). Base on budget and people interested to sponsor from greenYng we could invest some quantity, so how many people [...]

Please advise, how I can contribute to sponsoring this platform support. would like to target the following profile:
Connected to:
ESP32-S3
Features WiFi, BLE
Flash size 16MB GD25Q128 from GIGADEVICE (manufacturer 0x200 device 0x16408)
PSRAM: undetermined
Crystal 40MHz

JORGEGO, 2023-04-16:
lets move on!!!

josesimoes, 2023-04-16:
@mzirkle44060 thanks for considering sponsoring this feature.
You can either sponsor the project through our [Opencollective](https://opencollective.com/nanoframework) or ping me directly if you prefer being invoiced (for accounting or other company policies).

JORGEGO, 2023-04-16:
@mzirkle44060 and @josesimoes
From greenYng we need:

- to close a budget form @josesimoes 
- understand how do @mzirkle44060  will invest (50%) and 50%(greenYng)?

We understand the project is develop the firmware for  ESP32-S3FN8 (M5Stack in my case also)
we need to develop (https://shop.m5stack.com/collections/m5-controllers/products/m5stamp-esp32s3-module)

Right?

josesimoes, 2023-05-29:
The amount of work required to have this available has dropped significantly. We are now down to add support for S3 in nanoff!

If you're still interested in sponsoring this, now it's a good time as we'll have an empty slot towards the end of this week to work on this.

clau-bucur, 2023-05-29:
Sponsored too now a small amount for this!

dfch, 2023-07-06:
@josesimoes Hi there, I am also interested in having nano support ESP32-S3 and would like to sponsor (with invoice preferred). How can I help?

josesimoes, 2023-07-06:
@dfch appreciate that you're willing to contribute to the effort! 
Let me email you about this.

For the record: besides the work on nanoff it turned out that there were still some lose ends on the CLR and build system that required more work. We're currently working on these.

dfch, 2023-07-07:
@josesimoes got your email and made suggestions. Besides, good to hear you are already working on it.

aolszowka, 2023-07-07:
> For the record: besides the work on nanoff it turned out that there were still some lose ends on the CLR and build system that required more work. We're currently working on these.

@josesimoes Is there a milestone tracker for remaining outstanding issues we can track?

josesimoes, 2023-07-07:
@aolszowka there isn't. This issue is the one being use to track/report about this feature.

josesimoes, 2023-07-14:
Closing this one as the feature has been delivered.

closed 2023-07-14 by josesimoes

paule96, 2023-07-15:
@josesimoes it seems like this feature is only complete for the flasher. But the visual studio extension has still problems detecting the device.
I open earlier the issue #1327 for that.
Is this already tracked somewhere?

josesimoes, 2023-07-15:
@paule96 there is nothing specific for each device on the extension... The comms and detection algorithm is the same for  all targets.
What exactly are you seeing? Suggest that you ask about this in the Discord server for a more speedy interaction.

paule96, 2023-07-15:
I still see this very often:
[image]

Sadly I don't use discord

Alex-111, 2023-07-25:
It seems that no breakpoints are hit on a ESP32-S3-DevKitC-1 device.

Is this a bug?

[image]

Here are some more details:

Visual Studio -> 17.6.5
Visual Studio nanoframework Extension -> 2022.3.0.60

System Information
HAL build info: nanoCLR running @ ESP32_S3 built with ESP-IDF v4.4.5
  Target:   ESP32_S3
  Platform: ESP32

Firmware build Info:
  Date:        Jul 14 2023
  Type:        MinSizeRel build, support for PSRAM
  CLR Version: 1.8.1.328
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  NFApp1, 1.0.0.0
  nanoFramework.Hardware.Esp32, 1.6.3.0
  nanoFramework.Runtime.Events, 1.11.6.0
  System.Device.Gpio, 1.1.28.0
  mscorlib, 1.14.3.0

Native Assemblies:
  mscorlib v100.5.0.18, checksum 0xCCE8376E
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.9.0, checksum 0xC0E1540A
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3fcaaac0  0x00025000
  FLASH 0x00000000  0x00800000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x2E0000     Deployment
      2    0x00490000       1      0x200000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x00490000    0x200000 (2048kB)   Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x2E0000 (2944kB)   Deployment

Deployment Map
Empty

Alex-111, 2023-07-25:
Thanks for this quick response.
Shouldn‘t this ticket be er-opened, then?
I also noticed that the Wifi Sample code dies not work, too, i.e. There will be some work to do before we have full ESP-S3 support?

josesimoes, 2023-08-02:
> Thanks for this quick response. Shouldn‘t this ticket be er-opened, then? 

The support for ESP32S3 has been delivered, that's what this issue is about. 😉 
If there is an issue with _another_ matter it should be tracked on it's own.
Anyway WiFi is working now on S3.

josesimoes, 2023-08-02:
@romankiss that exception is thrown if: the operation is being performed with the wrong network interface type, if WiFi is not started nor initialized. I suspect that there isn't enough RAM available for IDF, which will cause the latter two. This is being investigated.

Alex-111, 2023-08-02:
@josesimoes WIFI seems not to work with ESP32-S3-BLE firmware on ESP32-S3-DevkitC-1. It works with ESP32-S3 firmware...

romankiss, 2023-08-02:
@josesimoes thank you for your response. I am using an _AtomS3 Lite_ product with firmware **AtomS3-1.8.1.404**, where the WiFi doesn't work, but when I switched to the firmware  **ESP32_S3-1.8.1.375** the WiFi is working very well.

josesimoes, 2023-08-02:
@romankiss nice catch! The target name was totally wrong on the cmake preset. 🤦🏻‍♂️ 🙄 
Now fixed and going through the pipeline.

romankiss, 2023-08-03:
@josesimoes  It looks like, the problem with WiFi is still in the firmware version **AtomS3-1.8.1.414**, the following is an exception:
    
    ++++ Exception System.InvalidOperationException - CLR_E_INVALID_OPERATION (1) ++++
    ++++ Message: 
    ++++ System.Device.Wifi.WifiAdapter::NativeScanAsync [IP: 0000] ++++
    ++++ System.Device.Wifi.WifiAdapter::ScanAsync [IP: 000d] ++++
    ++++ nanoFramework.Networking.WifiNetworkHelper::ScanAndConnectDhcp [IP: 000d] ++++

josesimoes, 2023-08-03:
@romankiss sorry to hear that. Care to raise another issue to properly track this specific problem?

FreeFrags, 2023-10-03:
Does this also support the LillyGo T-Display s3?

josesimoes, 2023-10-04:
> Does this also support the LillyGo T-Display s3?

It should. Although the display part is not enabled on the "stock" firmware.