nanoframework/Home #972: Migrate DS18B20 from nf-Community-Contribution to IoT.Device

Ellerbach, 2022-02-18:
Some rework needs to be done to align with quality standards.
Binding location: https://github.com/nanoframework/nf-Community-Contributions/tree/master/drivers/RFID_RDM6300_UART

Ellerbach, 2022-02-22:
> I can do it but please specify your expectations towards supported features: should be moved with all existing features like notification and alarms or just measurement on demand?

Great question, expectation is to have as many things moved. Especially anything that is already implemented in the code. It's not required to add more elements if they are not present.

There are also the Temperature to use UnitsNet and anything else that has units.

Elements on naming convention (like the class name, variables), code guidelines (extra enter between functions), unfortunately, we can't use *yet* Stylecop on the IoT Device repo. Once that in place, PR review on code style will be much easier. In the mean time, please be careful to those things.

And thanks a lot for proposition to do it!

rkalwak, 2022-02-22:
Ok then, I will do my best :)

rkalwak, 2022-08-28:
Done in PR: https://github.com/nanoframework/nanoFramework.IoT.Device/pull/526

closed 2022-08-29 by josesimoes

Ellerbach, 2022-08-29:
@rkalwak would you mind making a PR as well on the community repo to remove the implementation from there? So it's clean as well on that side.