nanoframework/Home #978: ESP_WROVER_KIT v4.1 Display Driver not working

KlausVcb, 2022-02-23:
### Target name(s)

ESP_WROVER_KIT

### Firmware version

1.7.4.285

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4
  Target:   ESP_WROVER_KIT
  Platform: ESP32

Firmware build Info:
  Date:        Feb 18 2022
  Type:        MinSizeRel build, chip rev. >= 0, support for PSRAM
  CLR Version: 1.7.4.43
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Graphics v100.0.0.3, checksum 0xB3EDAF22
  nanoFramework.Devices.OneWire v100.0.3.4, checksum 0xA5C172BD
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.0.1, checksum 0x5A31313D
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.1, checksum 0xA01012C3
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.3.0, checksum 0x9DB7198A
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.0.0, checksum 0x48031DC5
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3f80086c  0x003b0000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

The WROVER firmware includes the ILI9341 display driver to support the LCD out of the box
Trying to get the screen to work on an ESP_WROVER_KIT v4.1 fails while pins have been configured correctly according to the specs https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-wrover-kit.html :
- backLightPin = 5
- chipSelect = 22
- dataCommand = 21
- reset = 18
- SPI1_MISO = 25
- SPI1_MOSI = 23
- SPI1_CLOCK = 19

It is currently unclear if the firmware works on a V3 which was the case with earlier versions of the firmware (unknown to me which one) and hence if the reason is the version of WROVER (4.1)

There are alleged working examples on V4.1 + Python : https://gist.github.com/kumekay/11f1a3afa0fbafe8df2ac741a95ecc44

### How to reproduce

The project found [here](https://github.com/KlausVcb/nanoFramework/tree/main/src/WroverDisplayTest) can be used to reproduce the issue.
The screen lights up when executing but remains blank.
No text whatsoever appears. Other tests with drawing lines or shapes or using different coordinates yield the same results.

### Expected behaviour

The above mentioned project is supposed to write the current time to the screen.

### Aditional information

Sample project: https://github.com/KlausVcb/nanoFramework/tree/main/src/WroverDisplayTest

AdrianSoundy, 2022-02-23:
I can confirm this also doesn't work with Wrover V3 using ST7789V_240x320_SPI display either
Display lights up but Clear/Flush doesn't work.

Ellerbach, 2022-02-24:
It used to work when I added all the other drivers. The Wrover kit was the one I always tested all the changes. So I guess older *should* work. Now if anyone can test on an old version, after the addition of the new drivers and the initialization thru managed code, then we should be able to see any other change made in the mean time.

KlausVcb, 2022-02-24:
I flashed the WROOM v4.1 to firmware 1.7.0.2 which dates back to 25/8/2021 and adapted the sample solution to Core v1.10.5 and Graphics to v1.0.0 here:
https://github.com/KlausVcb/nanoFramework/tree/TestWithGraphicsV1.0.0/src/WroverDisplayTest

It works as intended!

I presume the change is indeed related to the introduction of managed initialization in October 21. When looking at https://github.com/nanoframework/nf-interpreter/blob/develop/src/nanoFramework.Graphics/Graphics/Displays/ILI9341_240x320_SPI.cpp this was done in commit https://github.com/nanoframework/nf-interpreter/commit/1245e8e806059128fac6551939e4bf4343b0ec61

AdrianSoundy, 2022-02-24:
@Ellerbach  Did you just test on m5stack as i know the m5stack was not working with original ILI9341 driver
I see some changes in the Initialization where done on that commit.

AdrianSoundy, 2022-02-24:
With the V3 wrover using ST7789 controller.
It clears display but screen is inverted and the orientation is wrong.
When i try to draw a line/flush its not appearing.

For the inversion i see a extra command for that is being sent. Removing that fixes it. Maybe that needs to be an screen option.

alberk8, 2022-02-25:
I have ST7789 working with some changes to native.  I will upload my working example shortly (one to two days).

Ellerbach, 2022-02-25:
> @Ellerbach Did you just test on m5stack as i know the m5stack was not working with original ILI9341 driver

I tested the changes with the Wrover and the various M5Stack. I may have miss something somewhere and had to make a lot of changes.

> I have ST7789 working with some changes to native. I will upload my working example shortly (one to two days).

Please, yes publish your work somewhere. 

I don't have anything to test currently. That may be a point where we will have to surface the orientation and the color scheme toove forward. I don't remember making a change in this driver but I may have unfortunately. And if that is the case apology for kot testing it on the Wrover.

To surface the color scheme should be straight forward but need to be added to all drivers.

Orientation is more challenging as it needs to be tested. I remember someone in our Discord who made it working and say he will go for PR. Will try to find the thread and ping some of you there.

josesimoes, 2023-01-16:
@Ellerbach when you have a few minutes, any chance you can look into this? It's open for quite sometime... 😉

Ellerbach, 2023-04-18:
So finally spending some time on this. I can confirm, the driver is not properly working. Neither the generic one. So investigating to find the root cause. Seems more an issue with the initialization than the rest of the logic. Interestingly enough, recently, a community member reported that the generic driver worked for him.

Ellerbach, 2023-04-18:
So made another test with the `Ili9341.GraphicDriverWithDefaultManufacturingSettings` generic graphic driver and it now works. So indeed, it's related to a missing or miss configured element somewhere. Investigating.

Ellerbach, 2023-04-18:
Fixed and in PR. Also while I'm there, will play a bit with the different orientations. With the generic driver, it's just so easy to test all this!

closed 2023-04-18 by Ellerbach