nanoframework/Home #981: Add Central/Client capability to Bluetooth

alberk8, 2022-02-27:
### Description

The current Bluetooth Library does not support connecting to peripheral Bluetooth device/sensor.

### How to solve the problem

Implement Client/Central role capability in Bluetooth Library

AdrianSoundy, 2022-03-01:
This is in progress.
Currently working on the BluetoothLEAdvertisementWatcher class

alberk8, 2022-03-02:
> This is in progress. Currently working on the BluetoothLEAdvertisementWatcher class
> 
> Still have to decide how to control which role the code is running Peripheral or Central. Maybe if you start the Peripheral/service advertising then in Peripheral /server role and Central tasks will give error.

I guess for a start, when one mode is running the other should be prevented to run.  I know that both can run simultaneously but on ESP32 it will be a stretch to get it running stable.

closed 2022-08-20 by AdrianSoundy