nanoframework/Home #983: System.Net.Http.HttpClient ssl get request throws exceptions to some but not all webhosts

KlausVcb, 2022-03-01:
### Library/API/IoT binding

System.Net.Http

### Visual Studio version

VS2022 17.2

### .NET nanoFramework extension version

2022.1.0.21

### Target name(s)

ESP_WROVER_KIT

### Firmware version

1.7.4.58

### Description

Using the HttpClient after setup of a wifi connection works well with http get requests but fails for https requests with a CLR_E_Fail SocketException.

### How to reproduce

1. Run the test project found [here](https://github.com/KlausVcb/nanoFramework/tree/main/src/HttpClientTest)
2. Replace the ssid and wifi pwd
3. When run: 
- the first request to [google.com](https://www.google.com/) with the Google wildcard certificate fails with a SocketException
- the second request to [google.com](https://www.google.com/) with the GlobalSign root certificate fails with an ArgumentOutOfRangeException
- the next two requests to [digicert.com](https://global-root-ca.chain-demos.digicert.com/) succeed

### Sample project or code

https://github.com/KlausVcb/nanoFramework/tree/main/src/HttpClientTest

alberk8, 2022-03-02:
can confirm that the GlobalSign RootCA works with HttpWebRequest, so there is something that is not right with HttpClient.  I could be related to buffer size??

Error
```
The thread '<No Name>' (0x2) has exited with code 0 (0x0).
    ++++ Exception System.ArgumentOutOfRangeException - 0x00000000 (1) ++++
    ++++ Message: 
    ++++ System.Net.Http.StreamContent::.ctor [IP: 0018] ++++
    ++++ System.Net.Http.HttpClientHandler::CreateResponseMessage [IP: 0021] ++++
    ++++ System.Net.Http.HttpClientHandler::Send [IP: 00d3] ++++
    ++++ System.Net.Http.HttpMessageInvoker::Send [IP: 0013] ++++
    ++++ System.Net.Http.HttpClient::Get [IP: 000c] ++++
    ++++ System.Net.Http.HttpClient::Send [IP: 006f] ++++
Exception thrown: 'System.ArgumentOutOfRangeException' in System.Net.Http.dll
An unhandled exception of type 'System.ArgumentOutOfRangeException' occurred in System.Net.Http.dll
```

CODE
```
  string url = "https://www.google.com";

var rootca = new X509Certificate(@"-----BEGIN CERTIFICATE-----
MIIDdTCCAl2gAwIBAgILBAAAAAABFUtaw5QwDQYJKoZIhvcNAQEFBQAwVzELMAkG
A1UEBhMCQkUxGTAXBgNVBAoTEEdsb2JhbFNpZ24gbnYtc2ExEDAOBgNVBAsTB1Jv
b3QgQ0ExGzAZBgNVBAMTEkdsb2JhbFNpZ24gUm9vdCBDQTAeFw05ODA5MDExMjAw
MDBaFw0yODAxMjgxMjAwMDBaMFcxCzAJBgNVBAYTAkJFMRkwFwYDVQQKExBHbG9i
YWxTaWduIG52LXNhMRAwDgYDVQQLEwdSb290IENBMRswGQYDVQQDExJHbG9iYWxT
aWduIFJvb3QgQ0EwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDaDuaZ
jc6j40+Kfvvxi4Mla+pIH/EqsLmVEQS98GPR4mdmzxzdzxtIK+6NiY6arymAZavp
xy0Sy6scTHAHoT0KMM0VjU/43dSMUBUc71DuxC73/OlS8pF94G3VNTCOXkNz8kHp
1Wrjsok6Vjk4bwY8iGlbKk3Fp1S4bInMm/k8yuX9ifUSPJJ4ltbcdG6TRGHRjcdG
snUOhugZitVtbNV4FpWi6cgKOOvyJBNPc1STE4U6G7weNLWLBYy5d4ux2x8gkasJ
U26Qzns3dLlwR5EiUWMWea6xrkEmCMgZK9FGqkjWZCrXgzT/LCrBbBlDSgeF59N8
9iFo7+ryUp9/k5DPAgMBAAGjQjBAMA4GA1UdDwEB/wQEAwIBBjAPBgNVHRMBAf8E
BTADAQH/MB0GA1UdDgQWBBRge2YaRQ2XyolQL30EzTSo//z9SzANBgkqhkiG9w0B
AQUFAAOCAQEA1nPnfE920I2/7LqivjTFKDK1fPxsnCwrvQmeU79rXqoRSLblCKOz
yj1hTdNGCbM+w6DjY1Ub8rrvrTnhQ7k4o+YviiY776BQVvnGCv04zcQLcFGUl5gE
38NflNUVyRRBnMRddWQVDf9VMOyGj/8N7yy5Y0b2qvzfvGn9LhJIZJrglfCm7ymP
AbEVtQwdpf5pLGkkeB6zpxxxYu7KyJesF12KwvhHhm4qxFYxldBniYUr+WymXUad
DKqC5JlR3XC321Y9YeRq4VzW9v493kHMB65jUr9TU/Qr6cf9tveCX4XSQRjbgbME
HMUfpIBvFSDJ3gyICh3WZlXi/EjJKSZp4A==
-----END CERTIFICATE-----");

            using var httpWebRequest = (HttpWebRequest)WebRequest.Create(new Uri(url));

            httpWebRequest.Method = "GET";
            httpWebRequest.SslProtocols = System.Net.Security.SslProtocols.Tls12;
            httpWebRequest.HttpsAuthentCert = rootca; 

            Debug.WriteLine(httpWebRequest.RequestUri.Host);
            Debug.WriteLine(httpWebRequest.RequestUri.AbsolutePath);
            try
            {
                using (var httpWebResponse = (HttpWebResponse)httpWebRequest.GetResponse())
                {

                    // wrap the response stream on a using statement to make sure that it's disposed
                    using (var stream = httpWebResponse.GetResponseStream())
                    {

                        // read response in chunks of 1k

                        byte[] buffer = new byte[1024];
                        int bytesRead = 0;
                        //int streamLength = (int)stream.Length;

                        Debug.WriteLine("Http response follows");
                        Debug.WriteLine(">>>>>>>>>>>>>");

                        do
                        {
                            bytesRead = stream.Read(buffer, 0, buffer.Length);
                            Debug.Write(Encoding.UTF8.GetString(buffer, 0, bytesRead));
                        }
                        while (bytesRead >= buffer.Length);
                    }

                    Debug.WriteLine(">>>>>>>>>>>>>");
                    Debug.WriteLine("End of Http response");

                    //Debug.WriteLine($"Internal Free Memory {totalFree}");
                }
            }
            catch (Exception ex)
            {
                Debug.WriteLine("\n#######################\n");
                Debug.WriteLine(ex.Message);
                Debug.WriteLine(ex.InnerException.Message);
            }
```

josesimoes, 2022-03-02:
@KlausVcb couple of comments on the app, for future reference.

1. Suggest that you use `WiFiNetworkHelper` instead of dealing directly with the `WiFiAdapter`. Much cleaner and works great.
2. The root CA that you're using there in GetGoogleCertificate is definitely not good. The one from GetGlobalSignCertificate it is.

closed 2022-03-02 by josesimoes