nanoframework/Home #984: HttpClient infinite loop after reading final chunk (length == 0)

KlausVcb, 2022-03-02:
### Library/API/IoT binding

System.Net.Http

### Visual Studio version

VS2022 17.2

### Description

When a chunked response is received the structure is as follows
- chunk with the lenght of the next chunk
- chunk with the data
- repeated x times until end of data
- last chunk with length == 0

HttpClient (or more precisely StreamContent) goes into an infinite loop when a chunk with length 0 is read.

### How to reproduce

Use HttpClient to call a webhost that returns a chunked response, for instance [this one](https://logicontro.azurewebsites.net/api/device/xyz/checksum) which returns 3 chunks:

6
AABEDT
0

josesimoes, 2022-03-02:
@KlausVcb seem that you've pinpoint the issue and fix shouldn't be that hard. Can you submit a PR with a fix? 😉

KlausVcb, 2022-03-02:
Yep, got it fixed already. Trying to figure out how to push a commit / PR ;-)

josesimoes, 2022-03-02:
Nice! Maybe our [contribution doc](https://docs.nanoframework.net/content/contributing/contributing-workflow.html#suggested-workflow) can help.

KlausVcb, 2022-03-02:
After a first (easy) attempt to fix this which was a no-go, I went deeper into the code and believe to have uncovered a bigger issue.

StreamContent is initialized with the stream.Length which at that point is an InputNetworkStreamWrapper clone of the NetworkStream. However, the length of that stream only reflects the data that is available at that point in time. So, when more data is flowing in on the socket, StreamContent stops reading when the initial bytes are read. I tested this with a download of a 5MB file which stopped reading after 3*2KB as the InputBuffer was initialized at 5139 bytes.

In my opinion the InputBuffer needs to be set to what is in the Content-Length header. This will at least solve it for non chunked responses.

Chunked responses on the other hand do not include a Content-Length header. The goal there is that a server can start to send data before the entire content is processed and hence the length is not known yet. In the current implementation the InputBuffer is initialized with what is already present on the stream. This includes the bytes with the length and potentially the zero length chunk to indicate the end of the data. The returned bytes read from InputNetworkStreamWrapper only reflect the actual chunk data however. So that is the root cause that the bytes read never surpass the inputbuffer length. Along with the current InputBuffer length issue, the actual length of the data is unknown. StreamContent should hence continue to read until it is clear that all data is processed. This could be achieved by casting the stream in StreamContent to InputNetworkStreamWrapper and checking a property or if the current decoupling is to be maintained (which I would choose personally) can be achieved by implementing an interface.

In conclusion, I presume a fix can be:
- Instantiate StreamContent with the length found in Content-Length header.
- If no such header is present (i.e. chunked response), implement an interface on InputNetworkStreamWrapper which can be checked by StreamContent to detect end of data.

I think I can introduce these changes but would like a first opinion before making an attempt. Just to verify this is indeed a viable approach.

josesimoes, 2022-03-02:
I agree with your assessment of the issue and the proposed fix.
Before nanoframework/System.Net.Http#294 (and pretty much inline with your reasoning above, which I've used in the initial implementation) Content-Lenght header was being used to set the stream length. 

When testing you Google sample on that issue, I've noticed that Content-Lenght header was showing -1, which is obviously invalid. On all my previous tests that value was there, so I assumed it could be safely used.
Yes, we have to take into account chunked data as we are supporting that.

So please go ahead with your proposed solution. Unfortunately, because of lack of network support we can't have Unit Test, otherwise those would be very helpful in all this.

KlausVcb, 2022-03-03:
The fix is implemented.

There is one other thing of note though. When testing the library with various urls, a request to https://www.google.com failed when using HttpContent.ReadAsString. Since now the StreamContent reads until the end, over 50kB was retrieved when getting google.com. The received data however contains characters which throws an exception (if I'm not mistaken) [here](https://github.com/nanoframework/nf-interpreter/blob/1d7aea72e4619f14f685ac67b4ceb31437e48b37/src/nanoFramework.System.Text/nf_system_text_System_Text_UTF8Encoding.cpp#L54). As example in this test there was 0x235 in the response at position 41k+ which is invalid in UTF8. Nevertheless, the response from google does denote <meta content="text/html; charset=UTF-8" http-equiv="Content-Type"> so 0x235 should not be in there.

Maybe it is nothing to worry about since querying webpages from NF is probably kinda out of scope :-)

If a solution is needed, perhaps an overload on UTF8Encoding.GetString which allows to ignore invalid bytes through a boolean?

josesimoes, 2022-03-03:
@KlausVcb thanks, I'll be reviewing it asap.

And regarding that hiccup when calling Google, I tottaly agree with you: our usage context here is not web browsing, so we should be good with UTF8 and what we have now. 😅

closed 2022-03-03 by josesimoes