nanoframework/Home #989: Implicit ToString() it's not working on some types

josesimoes, 2022-03-11:
### Details about Problem

**nanoFramework area:** (mscolib)

### Description
- When working in nanoframework/CoreLibrary#177 noticed that when running the Reflection sample, that implicit calls to ToString are not working for some types. Outputting the content of the DataRowAttribute fails for strings and bools. (~line 64 of Reflection\CustomAttributes\Program.cs).

jonstelly, 2022-04-13:
For Enums their `ToString()` returns their integer as a string.
`Console.WriteLine($"{DayOfWeek.Monday} - {DayOfWeek.Monday.ToString()}");`
outputs
`1 - 1`

josesimoes, 2022-04-13:
> For Enums their `ToString()` returns their integer as a string.
> `Console.WriteLine($"{DayOfWeek.Monday} - {DayOfWeek.Monday.ToString()}");`
> outputs
> `1 - 1`

https://docs.nanoframework.net/content/architecture/simplifications-and-trade-offs.html#enums

closed 2022-05-27 by josesimoes