nanoframework/Home #991: Cannot run.

bbhxwl, 2022-03-11:
### Tool

Visual Studio extension

### Description

Several people saw that the configuration was ok, but it just couldn't run.

### Screenshots

[image]
[image]

### Aditional context

[image]
[image]

Ellerbach, 2022-03-11:
Please update the CoreLibrary nuget, it will then work.

bbhxwl, 2022-03-11:
> Please update the CoreLibrary nuget, it will then work and add --preview in the nanoff command

https://discord.com/channels/478725473862549535/481782754674212867
Make sure it's a stable version bug, and the preview version can. Hope to repair

alberk8, 2022-03-14:
- Flash the stable ESP32 firmware 
`nanoff --target ESP32_REV0 --update --serialport COM3`

- Create a Hello World project and deploy.

**Console Output**
`16:51:17.355 [NanoDevices: valid device ESP32 @ COM6]
16:51:17.355 [NanoDevices: Serial device enumeration completed. Found 1 devices]
16:51:19.448 [Starting deployment transaction from v2019.10.0+f77bf3ee85]
16:51:19.450 [Starting debug engine on nanoDevice]
16:51:19.458 [Connecting to debug engine...OK]
16:51:19.946 [Couldn't find a valid native assembly required by mscorlib v1.11.7.2, checksum 0xC5322585]
16:51:19.946 [The connected target does not have support for mscorlib.]`

josesimoes, 2022-03-14:
There is no direct match between these two versions. The stable versions will get in sync in the next stable release of both mscorlib and firmware packages.

closed 2022-03-14 by josesimoes