nanoframework/Home #999: `nanoFramework.IoT.Device` has a mixture of namespace (`IoT` and `Iot`)

networkfusion, 2022-03-18:
### Library/API/IoT binding

`nanoFramework.IoT.Device`

### Visual Studio version

N/A

### .NET nanoFramework extension version

N/A

### Target name(s)

N/A

### Firmware version

N/A

### Device capabilities

N/A

### Description

`nanoFramework.IoT.Device` has a mixture of namespace (`IoT` and `Iot`). This "should" be aligned.

Needs to be fixed before a `stable` nuget is released.

### Expected behaviour

The same underlying namespace for all `Device` drivers.

josesimoes, 2022-03-25:
Hasn't this been fixed in a recent PR? 🤔

Ellerbach, 2022-03-25:
Fixed with https://github.com/nanoframework/nanoFramework.IoT.Device/pull/327

closed 2022-03-25 by Ellerbach