nanoframework/Home #1001: Inspect MSBuild extension vunrabilities

networkfusion, 2022-03-24:
### Details about Problem

https://github.com/nanoframework/nf-tools/security/dependabot

**nanoFramework area:** ( Visual Studio extension | Other tools )
[image]

### Description

See if we can update `karma` to `6.3.16` or above without causing issues.

josesimoes, 2022-03-24:
Those are security related vulnerabilities! 😯 

There's a reason why the URL you've put above it's only accessible to repo owners: it's supposed to be kept secret and not publicizing the vulnerability so it's not exploited. I've removed the image for this reason.

josesimoes, 2022-03-25:
@networkfusion dependencies updated. Thanks for the tip.

closed 2022-03-25 by josesimoes