nanoframework/Home #1002: nanoFramework.CoreLibrary 1.12.0-preview.19' is not compatible with 'nanoFramework.Graphics 1.0.2-preview.18

JORGEGO, 2022-03-24:
### Library/API/IoT binding

nanoFramework.Graphics 1.0.2-preview.18

### Description

Attempting to gather dependency information for package 'nanoFramework.M5Core.1.0.1-preview.64' with respect to project 'greenyng.device.nanoframework.common.m5basic', targeting '.NETnanoFramework,Version=v1.0'
Gathering dependency information took 110 ms
Attempting to resolve dependencies for package 'nanoFramework.M5Core.1.0.1-preview.64' with DependencyBehavior 'Lowest'
Unable to resolve dependencies. 'nanoFramework.CoreLibrary 1.12.0-preview.19' is not compatible with 'nanoFramework.Graphics 1.0.2-preview.18 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Hardware.Esp32 1.3.5-preview.10 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Iot.Device.Bmm150 1.0.288-preview.106 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Iot.Device.Button 1.0.288-preview.99 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Iot.Device.Buzzer 1.0.288-preview.103 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Iot.Device.Ip5306 1.0.288-preview.103 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Iot.Device.Mpu6886 1.0.288-preview.106 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.M5Core 1.0.1-preview.64 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.ResourceManager 1.1.4-preview.13 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Runtime.Events 1.10.0-preview.13 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Device.Gpio 1.0.3-preview.18 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Device.I2c 1.0.3-preview.13 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Device.Spi 1.0.4-preview.11 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Diagnostics.Stopwatch 1.0.289-preview.58 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.IO.Ports 1.0.3-preview.39 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.IO.Streams 1.0.0-preview.19 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Math 1.4.4-preview.22 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Numerics 1.0.289-preview.1 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Text 1.1.3-preview.24 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'UnitsNet.nanoFramework.Frequency 5.0.0-alpha005 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'UnitsNet.nanoFramework.Temperature 5.0.0-alpha005 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)'.
Time Elapsed: 00:00:00.1187638

Ellerbach, 2022-03-24:
See #1003

JORGEGO, 2022-03-24:
Thanks for the suggestions... I have uninstalled all as you could see on the picture... but
[image]

That it is the message:
Attempting to gather dependency information for package 'nanoFramework.AtomLite.1.0.1-preview.64' with respect to project 'greenyng.device.nanoframework.common.atom', targeting '.NETnanoFramework,Version=v1.0'
Gathering dependency information took 3.87 sec
Attempting to resolve dependencies for package 'nanoFramework.AtomLite.1.0.1-preview.64' with DependencyBehavior 'Lowest'
Unable to find a version of 'nanoFramework.System.Drawing' that is compatible with 'nanoFramework.AtomLite 1.0.1-preview.64 constraint: nanoFramework.System.Drawing (>= 1.0.288-preview.99)'.
Time Elapsed: 00:00:03.9213235
========== Finished ==========

JORGEGO, 2022-03-24:
All versions over >= 60 not it is possible to install
If I install version 58, the problem it is not compatible with the last image...

closed 2022-03-24 by josesimoes