nanoframework/Home #1003: nanoFramework.CoreLibrary 1.12.0-preview.19' is not compatible with 'nanoFramework.AtomLite 1.0.1-preview.64

JORGEGO, 2022-03-24:
### Library/API/IoT binding

nanoFramework.CoreLibrary 1.12.0-preview.19

### Description

As you could see on the screenshot, I have lasted version of  nanoFramework.CoreLibrary,

but when I try to upgrade the version of Atom:

Attempting to gather dependency information for package 'nanoFramework.AtomLite.1.0.1-preview.64' with respect to project 'greenyng.device.nanoframework.common.atom', targeting '.NETnanoFramework,Version=v1.0'
Gathering dependency information took 41 ms
Attempting to resolve dependencies for package 'nanoFramework.AtomLite.1.0.1-preview.64' with DependencyBehavior 'Lowest'
Unable to resolve dependencies. 'nanoFramework.CoreLibrary 1.12.0-preview.19' is not compatible with 'nanoFramework.AtomLite 1.0.1-preview.64 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Hardware.Esp32 1.3.5-preview.10 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Hardware.Esp32.Rmt 1.2.2-preview.10 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Iot.Device.Button 1.0.288-preview.99 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.Runtime.Events 1.10.0-preview.13 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Device.Adc 1.0.2-preview.13 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Device.Dac 1.4.3-preview.13 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Device.Gpio 1.0.3-preview.18 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Device.I2c 1.0.3-preview.13 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Device.Spi 1.0.4-preview.11 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)', 'nanoFramework.System.Drawing 1.0.288-preview.97 constraint: nanoFramework.CoreLibrary (>= 1.12.0-preview.19)'.
Time Elapsed: 00:00:00.0977196
========== Finished ==========

### Screenshots

[image]

Ellerbach, 2022-03-24:
Please update the AtomLite nuget BEFORE updating anything else. By updating it all the dependencies will update properly.
I don't have a AtomLite but just did on Core2 and works as expected.
Other option: reverse back to the previous version those you have updated manually and then update the AtomLite.
Other option: remove all :-) Add the AtomLite and all the dependencies will come with it properly.

josesimoes, 2022-03-25:
Should be fixed with the latest release of M5Stack NuGets.

closed 2022-03-25 by josesimoes