nanoframework/Home #1009: VS2022: Devi tool window icons missing after VS extension update

networkfusion, 2022-04-06:
### Tool

Visual Studio extension

### Description

Currently the icons sometimes "disappear" when the VS2022 nanoFw extension is updated.
This is a potential solution is outlined in "https://stackoverflow.com/questions/58311920/visual-studio-vsix-extension-not-showing-in-tools-menu" (although there are possibily other fixes).
A solution like this "should" be tried before escalation to the VS team, as although there is a workaround (via: devenv /updateConfiguration) it is annoying.

### Screenshots

[image]

### Aditional context

A test fix was applied via https://github.com/nanoframework/nf-Visual-Studio-extension/pull/710 but it was unsuccessful.

PNPTransistor, 2022-04-16:
Hi,

Not that this helps - but I'm seeing the same thing. All NF buttons in solution explorer are now blank.

Ellerbach, 2022-04-17:
> Hi,
> 
> Not that this helps - but I'm seeing the same thing. All NF buttons in solution explorer are now blank.

This is how you can fix it, run: `devenv /updateConfiguration` from the developer command prompt. Close all the Visual Studio instances before.

oloendithas, 2022-09-21:
Indeed, deployment feature also restored by running `devenv /updateConfiguration`

DaveSchmid, 2022-12-18:
[image]

josesimoes, 2023-01-02:
## ⚠️ Issue reported to VS team. Please up-vote [here](https://developercommunity.visualstudio.com/t/Extension-icons-missing-after-update/10242633)!

josesimoes, 2023-01-16:
There are reports of this happening, so the fix may not be that good...
cc @DaveSchmid

josesimoes, 2023-01-17:
Instructions on how to gather a log of issues loading images, taken from [here](https://developercommunity.visualstudio.com/t/Extension-icons-missing-after-update/10242633#TPIN-N10245020-N10254729)

>>>>>>>>

To toggle logging, follow the instructions below. That will create a log file called `ImageLibrary.log` in the `%USERPROFILE%` folder. 

1. Enable the logging with
```
vsregedit set local HKLM “ImageLibrary” TraceLevel string Verbose
vsregedit set local HKLM “ImageLibrary” TraceFilename string ImageLibrary.log
```
2. With VS closed, open a VS developer prompt and run `devenv /updateConfiguration` to ensure any caching issues are captured
3. Then launch VS and ensure your icons are missing to capture any runtime issues. 
4. Grab the log file  `ImageLibrary.log` in the `%USERPROFILE%` folder. 
5. Turn off tracing with
```
vsregedit set local HKLM “ImageLibrary” TraceLevel string Off
```
6. Upload your log in this issue and/or directly at the issue in VS Dev Community [here](https://developercommunity.visualstudio.com/t/Extension-icons-missing-after-update/10242633#TPIN-N10245020-N10254729)

DaveSchmid, 2023-01-17:
I tried it. Now the icons are back. :- ) 
But no ImageLibrary.log on the whole computer.

josesimoes, 2023-03-09:
@TerryFogg do not upin this issue. It's critical and it's meant to be highlighted at the top.

TerryFogg, 2023-03-09:
@josesimoes  Sorry, I don't remember doing this? and I wouldn't normally. I will be a bit more careful

matsujirushi, 2023-03-09:
I did this steps, but VS didn't generate ImageLibrary.log file. 😢
https://github.com/nanoframework/Home/issues/1009#issuecomment-1385553170

[image]

[image]

atboxshop, 2023-03-24:
So is there any update to solve this problem so far guys?

SoaSoftNet, 2023-04-13:
Thank you for your hard work. Please let me know when you repair it. Thank you.

josesimoes, 2023-04-13:
> Thank you for your hard work. 

Thank you for your kind words.

> Please let me know when you repair it. Thank you.

Just subscribe to the issue that's tracking this and you'll be notified.

pt2go, 2023-05-04:
update to vs2022 17.5.5，the icon coming back..............

matsujirushi, 2023-05-04:
@pt2go me too.

atboxshop, 2023-05-05:
After updating to version 17.5.5, the device explorer appear just 1 time :) and it was disappear after restart visual studio 2022. Pls fix it. Thanks guys.

pt2go, 2023-05-06:
> Still missing icons after updating to version 17.5.5, the device explorer's icons appear just 1 time :) right after the updating and its icons were disappear again after restart visual studio 2022. Pls fix it. Thanks guys.

.......right，icon miss

renerlemes, 2023-05-25:
I updated VS to 17.6.2 version and now is working

fhtino, 2023-05-27:
Updated to VS 17.6.2 and is **not** working  :(

[image]

pt2go, 2023-06-07:
> I updated VS to 17.6.2 version and now is working

my vs with 17.6.2 working perfect!

atboxshop, 2023-06-09:
After updated my nanoframework extension for visual studio to the latest version and updated my visual studio to version 17.6.2, the icon re-appear again and it works perfectly. Thanks guys. 👍

fhtino, 2023-06-09:
I confirm now it's working.  VS 17.6.2 and nanoFramework extension 2022.3.0.60 solved the issue.

jouda13, 2023-07-11:
Hi, I’ve had a bad luck with mine. VS 17.6.5 and nanoFramework extension 2022.3.0.60 . The icons are still missing. Anybody else is having the same issue?

Ellerbach, 2023-07-12:
> Hi, I’ve had a bad luck with mine. VS 17.6.5 and nanoFramework extension 2022.3.0.60 . The icons are still missing. Anybody else is having the same issue?

Yes, I did. Closing all Visual Studio, then openeing the VS Dev command prompt and then running: `devenv /updateConfiguration` did fix the issue.

atboxshop, 2023-07-13:
Confirm that after update vs studio 2022 to version 17.6.5, the icon was disappeared again. I also tried to run the devenv cmd for updating the configuration but could not fix the problem. Pls fix it guys. Thanks

atboxshop, 2023-07-15:
To fix the missing icon after updating vs studio 2022 to 17.6.5, I go to start -> all apps -> Visual Studio 2022 -> x64 native tool command prompt or x86 native tools command prompt . And run this command devenv /updateConfiguration to fix the problem like Ellerbach mention in the previous posting. :D

manuel76413, 2023-08-20:
> devenv /updateConfiguration

doesn't take effect, still blank toolbar, empty icon

atboxshop, 2023-08-21:
Run x64 native tool command prompt as administrator

josesimoes, 2023-09-01:
After updating to v17.7.3 a few days ago, I still have the extension with icons. Hopefully this has been fixed upstream...

torbacz, 2023-09-08:
One week later, icons have disappeared :(

atboxshop, 2023-11-14:
Any updates to fix this issue guys 👍 . My suggestion is that this device explorer window must remove and replace that window by a sub-menu somehow. 👍

TerryFogg, 2023-11-14:
It quite frustrating as it "comes and goes". About a month ago I lost my icons.
The command "devenv /updateconfiguration" run at the developer prompt can fix this sometimes.
I don't have icons at the moment, and this did not fix my problem.
I have been waiting for Microsoft to finish off their new Visual Studio Extension development system which is easier to debug and develop ( I hope). Still waiting and when it comes we could use this

TerryFogg, 2023-11-14:
I did wonder if the fact that our icons are .png and not the older .ico has anything to do with the mishandling by Microsoft code.

CoryCharlton, 2023-11-14:
Not sure if it helps at all but it looks like the issue also impacts the project icons (library and application projects but not unit tests):

[image]

josesimoes, 2023-11-14:
> Any updates to fix this issue guys 👍 . My suggestion is that this device explorer window must remove and replace that window by a sub-menu somehow. 👍

How's that? There is already an entry under View menu to show/hide Device Explorer.

josesimoes, 2023-11-14:
If you guys go through the related issue in VS Developer Community you'll see that this has to do with the naming of the resources, and the strong naming and other stuff that VS resource manager is doing trying to find the image resources.
Unrelated with image the format. And yes, it affect anything image resource related... 😞

atboxshop, 2023-11-17:
> > Any updates to fix this issue guys 👍 . My suggestion is that this device explorer window must remove and replace that window by a sub-menu somehow. 👍
> 
> How's that? There is already an entry under View menu to show/hide Device Explorer.
Yes, we can hide or show the device explorer. I means if you guys can create an sub menu (without icon or using default icon) for example it will be like this in Visual Studio : View-->Other Windows-->Device Explorer--> Ping Device or Show Setting. And problem will be solved. 👍

josesimoes, 2023-11-27:
Hey @softworkz! Just read your detailed analysis at developer community issue. Nice.
Thanks for taking the time to investigate this and provide a fix.
We'll be on this ASAP! 👍🏻 🥇

Ellerbach, 2023-11-27:
With 2023.3.0.67 all works fine again! Thanks a lot @softworkz for the deep investigation!

softworkz, 2023-11-28:
The favor is mutual since I learn a lot from existing VS project system extensions. 

Let's help other extensions developers not needing to go through all the trouble by upvoting here:  https://developercommunity.visualstudio.com/t/VSSDKCPSExtensibility:-Image-Loading-i/10525678

and here: https://github.com/microsoft/VSProjectSystem/issues/373

networkfusion, 2023-12-20:
Re-opening as unfortunately it is still occuring.

atboxshop, 2023-12-26:
To fix it temporary, go to extension, disable nanoframework and re-enable it. After that restart your visual studio, all icons will appear.

networkfusion, 2023-12-27:
> To fix it temporary, go to extension, disable nanoframework and re-enable it. After that restart your visual studio, all icons will appear.

Nice workaround!

kondr1, 2024-02-28:
Both workaround doesn't work for me.

alberk8, 2024-02-28:
@kondr1,  Can you try to repair your Visual Studio installation?

madschristiansen, 2024-06-06:
This is a massive issue for newcomers to this project. I have now tried for days to get startet but the fact thatI cant use the vs extension, I cant change network setting (Or at least cant find any other way)
Is it better on older VS version like 2019?

I need it for a POC so I should have been done yesterday but have spend all my time installing/removing/diabling/enabling/running dev commands etc etc.

josesimoes, 2024-06-06:
@madschristiansen I'm totally sympathetic with you on how much this is a pain! 
As you can see from the thread above, this is an issue caused by a 3rd party tool (Visual Studio) of which we don't have any control. 
For some reason, the reports on this problem aren't getting enough love from the VS team. If you could go to the developer community website (URLs above) and add your comments there expressing the need to have this fixed, it would be helpful. Considering that issues there are prioritized according to the impact they are causing to developers.

softworkz, 2024-06-06:
I would recommend to create a support incident with MS about this. Refer to the issue on VS community and ask for escalation.

Three ways to create incidents:

- Either you have a VS subsccription with free incidents include
- you or your company has a support contract
- you can buy individual incidents (expensive, though)

If they don't resolve it, the incident won't be counted as used (but you don't get the money back for the incident package in the 3rd case, yet you can use them otherwise)

josesimoes, 2024-06-12:
For what is worth: after update VS to 17.10.1 and install the latest version of our extension I have the icons back. 🤞🏻 😅

mikmog, 2024-06-14:
Also in Version 17.11.0 Preview 2.0 😊

josesimoes, 2024-08-22:
Looks like this has been fixed by VS Team. Let's wait for the fix to become available.
See [comment](https://developercommunity.visualstudio.com/t/VSSDKCPSExtensibility:-Image-Loading-i/10525678#T-N10728516) in the follow-up issue.

zorgoz, 2024-08-25:
Visual Studio Version 17.12.0 Preview 1.0 (Released August 13th): no icons - neither in the View > Other Windows menu.  The fix referred by @josesimoes might not be included yet. Let's wait and see...

ajan0636, 2024-10-31:
still not working

josesimoes, 2024-11-15:
New issue raised in Developer Community: https://developercommunity.visualstudio.com/t/Image-icons-from-image-catalog-not-showi/10791720

closed 2024-11-22 by josesimoes

atboxshop, 2024-11-25:
After getting back all of icons. Now, my esp32 was not detected in device explorer tool. It was detected when I use command nandoff list devices. Try to update firmware 
 and repair/reinstall Visual Studio but still could not got my esp32 detecting in device explorer. Any suggestion guys? 
[image]
[image]