nanoframework/Home #1010: System.Net.NetworkInformation.IPGlobalProperties.GetIPAddress() returns "255.255.255.255" when running debugger.

networkfusion, 2022-04-07:
### Library/API/IoT binding

nanoFramework.Networking?

### Visual Studio version

VS2022

### .NET nanoFramework extension version

latest

### Target name(s)

OrgPalThree

### Firmware version

latest

### Device capabilities

``` System Information
HAL build info: nanoCLR running @ ORGPAL_PALTHREE built with ChibiOS v2021.11.1.11
  Target:   ORGPAL_PALTHREE
  Platform: STM32F7

Firmware build Info:
  Date:        Apr  1 2022
  Type:        MinSizeRel build with ChibiOS v2021.11.1.11
  CLR Version: 1.8.0.6
  Compiler:    GNU ARM GCC v10.3.1

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  000000000000000000004E46504B54563100000
  00001000E99E0012000

Target capabilities:
  Has nanoBooter: YES
  nanoBooter: v1.8.0.6
  IFU capable: NO
  Has proprietary bootloader: NO

AppDomains:

Assemblies:
  debugIpTest, 1.0.0.0
  System.Net, 1.8.1.3
  System.IO.Streams, 1.0.0.2
  mscorlib, 1.12.0.4
  System.Threading, 1.0.4.3
  nanoFramework.Runtime.Events, 1.10.0.3
  nanoFramework.System.Text, 1.1.3.3

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Stm32 v100.0.4.4, checksum 0x0874B6FE
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.1, checksum 0xA01012C3
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
[... 170 lines cut ...]
Waiting for network up and IP address...
    ++++ Exception System.InvalidOperationException - 0x00000000 (1) ++++
    ++++ Message: 
    ++++ nanoFramework.Networking.NetworkHelper::SetupHelper [IP: 0008] ++++
    ++++ nanoFramework.Networking.NetworkHelper::InternalWaitNetworkAvailable [IP: 0004] ++++
    ++++ nanoFramework.Networking.NetworkHelper::SetupAndConnectNetwork [IP: 0010] ++++
    ++++ nanoFramework.Networking.NetworkHelper::SetupAndConnectNetwork [IP: 0006] ++++
    ++++ debugIpTest.Program::SetupDeviceNetwork [IP: 001a] ++++
    ++++ debugIpTest.Program::Main [IP: 0010] ++++
Failed to receive an IP address and/or valid DateTime. Error: 7.
Failed to receive an IP address and/or valid DateTime. Error: System.InvalidOperationException: Exception was thrown: System.InvalidOperationException.
IP = 255.255.255.255
RTC = 01/01/1980 00:00:51
Waiting for network up and IP address...
    ++++ Exception System.InvalidOperationException - 0x00000000 (1) ++++
    ++++ Message: 
    ++++ nanoFramework.Networking.NetworkHelper::SetupHelper [IP: 0008] ++++
    ++++ nanoFramework.Networking.NetworkHelper::InternalWaitNetworkAvailable [IP: 0004] ++++
    ++++ nanoFramework.Networking.NetworkHelper::SetupAndConnectNetwork [IP: 0010] ++++
    ++++ nanoFramework.Networking.NetworkHelper::SetupAndConnectNetwork [IP: 0006] ++++
    ++++ debugIpTest.Program::SetupDeviceNetwork [IP: 001a] ++++
    ++++ debugIpTest.Program::Main [IP: 0010] ++++
Failed to receive an IP address and/or valid DateTime. Error: 7.
Failed to receive an IP address and/or valid DateTime. Error: System.InvalidOperationException: Exception was thrown: System.InvalidOperationException.
```

networkfusion, 2022-04-07:
Running the same code on an `STM32F769I` leads to the same result (in my environment).

networkfusion, 2022-04-12:
This does not seem to be a probelem on an `M5Core2` device, so think it might be `STM32` related only. OR related to Ethernet vs WiFi!

The program run on the `M5Core2` was:

```
// Licensed to the .NET Foundation under one or more agreements.
// The .NET Foundation licenses this file to you under the MIT license.

using nanoFramework.M5Core2;
using nanoFramework.M5Stack;
using nanoFramework.Networking;
using nanoFramework.Runtime.Native;
using System;
using System.Diagnostics;
using System.Threading;
using Console = nanoFramework.M5Stack.Console;

M5Core2.InitializeScreen();

Debug.WriteLine("Hello from M5Core2!");

// Give 60 seconds to the wifi join to happen
CancellationTokenSource cs = new(60000);
var success = WiFiNetworkHelper.ConnectDhcp(Secrets.Ssid, Secrets.Password, requiresDateTime: true, token: cs.Token);
if (!success)
{
    // Something went wrong, you can get details with the ConnectionError property:
    Debug.WriteLine($"Can't connect to the network, error: {WiFiNetworkHelper.Status}");
    if (WiFiNetworkHelper.HelperException != null)
    {
        Debug.WriteLine($"ex: {WiFiNetworkHelper.HelperException}");
    }
}

Debug.WriteLine("Network Setup complete.");
Debug.WriteLine($"IP = {System.Net.NetworkInformation.IPGlobalProperties.GetIPAddress()}");
Debug.WriteLine($"RTC = {DateTime.UtcNow}");

var mpu = M5Core2.AccelerometerGyroscope;

M5Core2.TouchEvent += TouchEventCallback;

Thread.Sleep(Timeout.Infinite);

void TouchEventCallback(object sender, TouchEventArgs e)
{
    const string StrLB = "LEFT BUTTON PRESSED  ";
    const string StrMB = "MIDDLE BUTTON PRESSED  ";
    const string StrRB = "RIGHT BUTTON PRESSED  ";
    const string StrXY1 = "TOUCHED at X= ";
    const string StrXY2 = ",Y= ";
    const string StrID = ",Id= ";
    const string StrDoubleTouch = "Double touch. ";
    const string StrMove = "Moving... ";
    const string StrLiftUp = "Lift up. ";

    Debug.WriteLine($"Touch Panel Event Received Category= {e.EventCategory} Subcategory= {e.TouchEventCategory}");
    Console.CursorLeft = 0;
    Console.CursorTop = 0;

    Debug.WriteLine(StrXY1 + e.X + StrXY2 + e.Y + StrID + e.Id);
    Console.WriteLine(StrXY1 + e.X + StrXY2 + e.Y + StrID + e.Id + "  ");

    if ((e.TouchEventCategory & TouchEventCategory.LeftButton) == TouchEventCategory.LeftButton)
    {
        Debug.WriteLine(StrLB);
        Console.WriteLine(StrLB);
    }
    else if ((e.TouchEventCategory & TouchEventCategory.MiddleButton) == TouchEventCategory.MiddleButton)
    {
        Debug.WriteLine(StrMB);
        Console.WriteLine(StrMB);
    }
    else if ((e.TouchEventCategory & TouchEventCategory.RightButton) == TouchEventCategory.RightButton)
    {
        Debug.WriteLine(StrRB);
        Console.WriteLine(StrRB);
    }

    if ((e.TouchEventCategory & TouchEventCategory.Moving) == TouchEventCategory.Moving)
    {
        Debug.WriteLine(StrMove);
        Console.Write(StrMove);
    }

    if ((e.TouchEventCategory & TouchEventCategory.LiftUp) == TouchEventCategory.LiftUp)
    {
        Debug.WriteLine(StrLiftUp);
        Console.Write(StrLiftUp);
    }

    if ((e.TouchEventCategory & TouchEventCategory.DoubleTouch) == TouchEventCategory.DoubleTouch)
    {
        Debug.WriteLine(StrDoubleTouch);
        Console.Write(StrDoubleTouch);
    }

    Console.WriteLine("                                    ");
    Debug.WriteLine($"IP = {System.Net.NetworkInformation.IPGlobalProperties.GetIPAddress()}");
    Debug.WriteLine($"RTC = {DateTime.UtcNow}");
    Console.WriteLine($"IP = {System.Net.NetworkInformation.IPGlobalProperties.GetIPAddress()}");
    Console.WriteLine($"RTC = {DateTime.UtcNow}");
    Console.WriteLine("                                    ");
    Console.WriteLine("                                    ");
    Console.WriteLine("                                    ");
}

```

and returned:
[image]

networkfusion, 2022-05-03:
Personally, this is still relateted to: System.Net.NetworkInformation.IPGlobalProperties.GetIPAddress() returns "255.255.255.255" when debugger is attached. When the device is running without the debugger, it is able to display the IP correctly. (although possibly needs a different issue, deployment is flaky at best.

josesimoes, 2022-05-26:
@networkfusion please confirm if you're still seeing this happening.

networkfusion, 2022-05-30:
I still have issues with SNTP when the debugger is NOT running, but that is a seperate bug: https://github.com/nanoframework/Home/issues/739

So closing this.

closed 2022-05-30 by networkfusion