nanoframework/Home #1012: Unable to flash ESP32 device and deploy (alternative method) on Linux

shodan8192, 2022-04-11:
### Tool

Visual Studio Code extension

### Description

After selecting target board and image version nothing happens. The problem is :

`spawnSync ~/.vscode/extensions/nanoframework.vscode-nanoframework-1.0.56/dist/utils/serial-monitor-cli/linux/main ENOENT`

on disk, that executable have .out extension. After removing it, port and baudrate can be chosen.

After selecting baudrate I get error :

```
Unhandled exception. System.ComponentModel.Win32Exception (13): An error occurred trying to start process '~/.vscode/extensions/nanoframework.vscode-nanoframework-1.0.56/dist/utils/nanoFirmwareFlasher/esptool/esptoolLinux/esptool'
with working directory '~/.vscode/extensions/nanoframework.vscode-nanoframework-1.0.56/dist/utils/nanoFirmwareFlasher/esptool/esptoolLinux'. Permission denied
```

It's turn out, that esptool doesn't have execute permission. After correcting this, firmware was flashed successfully - also problem with deploy by alternative method is solved

Additionally there is a redundant folder called 'esptooLinux' 

### Aditional context

OS : Pop!_OS 21.10

Mono : 6.12.0 Preview (6.12.0.147)
.NET Runtime : 6.0 (6.0.3-1)

VSCode : 1.66.1
.NET nanoFramework extension : 1.0.56

Ellerbach, 2022-04-15:
Can you try to give the executable permission to the linux esptool? That should be the issue there

shodan8192, 2022-04-15:
I've already described this as solution :

> It's turn out, that esptool doesn't have execute permission. After correcting this, firmware was flashed successfully - also problem with deploy by alternative method is solved

Ellerbach, 2022-04-15:
> I've already described this as solution :
> 
> > It's turn out, that esptool doesn't have execute permission. After correcting this, firmware was flashed successfully - also problem with deploy by alternative method is solved

ho, sorry, miss read... We'll try to solve this issue for the installation. thanks again!

josesimoes, 2022-04-29:
pinging @Ellerbach... when you have a few minutes can you look into this please?

Ellerbach, 2022-05-06:
@shodan8192 can you please clone the PR I just did and test if the fix is working? I have no native Linux machine close to me, so I cannot test it properly.
If that doesn't work, I have another idea to fix it. And same, I would love to have your help.

shodan8192, 2022-05-06:
Fix doesn't work.

`Unhandled exception. System.ComponentModel.Win32Exception (13): An error occurred trying to start process '/home/shodan/.vscode/extensions/nanoframework.vscode-nanoframework-1.0.72/dist/utils/nanoFirmwareFlasher/esptool/esptoolLinux/./esptool' with working directory '/home/shodan/.vscode/extensions/nanoframework.vscode-nanoframework-1.0.72/dist/utils/nanoFirmwareFlasher/esptool/esptoolLinux'. Permission denied`

next thing is that this line : https://github.com/nanoframework/nf-VSCodeExtension/blob/60b549fe0f592bbe9f396f46ac7f10a524ea4763/src/serialportctrl.ts#L46

should be
``` ts
 fileName = "main.out";
```

shodan8192, 2022-05-07:
I think that problem with permissions of esptool is elsewhere.

Please, look at the end of https://github.com/nanoframework/nf-VSCodeExtension/blob/main/scripts/build.ps1, where `chmod -R +x ./$outputDirectory/utils/` is called.

That's works fine if building locally, but when building in Azure Pipeline then nanoFirmwareFlasher build is skipped in `build.ps1` script, so `chmod` have no effect, because there's no esptool in dist/utils yet.

josesimoes, 2022-05-07:
@shodan8192 I haven't looked at the or the pipeline script. If you know what to change to fix this in the pipeline feel free to suggest the change right here in the PR. 😉

shodan8192, 2022-05-08:
OK, I'll try to make PR for this tomorrow

closed 2022-05-09 by josesimoes