nanoframework/Home #1013: Performance differences between 1.8.0.6 ESP32_REV3 and ESP32_PSRAM_REV3 firmware

shodan8192, 2022-04-11:
### Target name(s)

ESP32_PSRAM_REV3, ESP32_REV3

### Firmware version

1.8.0.6

### Description

``` cs
using System.Device.Gpio;

namespace Blinky
{
    public class Program
    {
        private static GpioController s_GpioController;
        public static void Main()
        {
            s_GpioController = new GpioController();
            GpioPin led = s_GpioController.OpenPin(25, PinMode.Output);

            while (true)
            {
                led.Toggle();
            }
        }
    }
}
```
Output frequency on pin 25 :
```
81kHz on ESP32_PSRAM_REV3
27kHz on ESP32_REV3
```

``` cs
using System.Device.Dac;

namespace Blinky
{
    public class Program
    {
        public static void Main()
        {
            DacController dac = DacController.GetDefault();
            DacChannel dacChannel = dac.OpenChannel(0);

            while (true)
            {
                dacChannel.WriteValue(0);
                dacChannel.WriteValue(255);
            }
        }
    }
}

```
Output frequency on pin 25 :
```
4.7kHz on ESP32_PSRAM_REV3
2.5kHz on ESP32_REV3
```
What's curious, that replacing led.Toggle() with led.Write(0) ; led.Write(1) gives :
```
1.5kHz on ESP32_PSRAM_REV3
1.7kHz on ESP32_REV3
```

Mandelbrot fractal calculations (https://gist.github.com/shodan8192/1248f49a1d752e34a7abc42a668b0b0d) shows little general performance difference
```
7390ms on ESP32_PSRAM_REV3
7530ms on ESP32_REV3
```

Ellerbach, 2022-04-12:
Thanks for sharing @shodan8192.
Few elements: Toggling the pin is always faster as there is a hardware way of doing it directly. All other methods goes a longer way.
For the PSRAM vs non PSRAM performance difference, hard to comment why.

And question: any expectation or anything that you feel is broken here?

shodan8192, 2022-04-12:
I'm just curious frome where this come from. It's more expected for me that the PSRAM build will be slightly slower.
While pin toggling it's not common use case, it's unknown to me where else that difference may shows up and why.