nanoframework/Home #1017: Race condition in nf-debugger

shodan8192, 2022-04-15:
### Tool

Visual Studio Code extension

### Description

There's a race condition in nf-debugger code : `_newDevicesCount` increment and decrement at lines below, should be lock-guarded

https://github.com/nanoframework/nf-debugger/blob/149bf81815d95b94fe7fe035fc5d8834671574d3/nanoFramework.Tools.DebugLibrary.Shared/PortSerial/PortSerialManager.cs#L430

https://github.com/nanoframework/nf-debugger/blob/149bf81815d95b94fe7fe035fc5d8834671574d3/nanoFramework.Tools.DebugLibrary.Shared/PortSerial/PortSerialManager.cs#L280

Without lock, nanoFrameworkDeployer wait for enumeration complete here :

https://github.com/nanoframework/nanoFrameworkDeployer/blob/fb37785a60186ad445bb856855c742e793df05e9/nanoFrameworkDeployer/Program.cs#L134

which may never happen, as `_newDevicesCount` may not reach value 0

### How to reproduce

Try nanoFramework : Deploy Project in vscode
Seems to be easily reproducible when there are many ports to enumerate eg on Linux machine, which has 32 hardcoded /dev/ttyS* nodes

### Aditional context

OS : Pop!_OS 21.10
Mono : 6.12.0 Preview (6.12.0.147)
.NET Runtime : 6.0 (6.0.3-1) 

VSCode : 1.66.1
.NET nanoFramework extension : 1.0.56
C# extension : 1.24.3

josesimoes, 2022-04-15:
Nice catch and it makes sense. I've seen this happening a few times. 😅

Can I interest you on submitting a PR with a fix? 😉

shodan8192, 2022-04-15:
There's another problem which I'm working on. I will open new issue, then prepare PR for both.

closed 2022-04-19 by josesimoes