nanoframework/Home #1018: Serial port opening by nf-debugger on Linux causing ESP32 board unintentional reset

shodan8192, 2022-04-15:
### Tool

Visual Studio Code extension

### Description

On Linux, calling PortSerial.OpenDevice() in nf-debugger causing momentary RTS and DTR assertion. RTS is held for ~140us longer than DTR, which triggers ESP32 reset, which in turn prevents finding it :
```
Finding valid ports
Found: 0 devices
Finding devices, attempt 1
Found: 0 devices
Finding devices, attempt 2
Found: 0 devices
Finding devices, attempt 3
Found: 0 devices
Finding devices, attempt 4
Found: 0 devices
Error too many retries
Exit with return code 1
```
To avoid this, it seems sufficient to explicitly assert DTR/RTS before port opening :
https://github.com/nanoframework/nf-debugger/blob/149bf81815d95b94fe7fe035fc5d8834671574d3/nanoFramework.Tools.DebugLibrary.Shared/PortSerial/PortSerial.cs#L81

``` cs
NanoDevice.DeviceBase = new SerialPort(InstanceId, BaudRate, Parity.None, 8);

Device.DtrEnable = true;
Device.RtsEnable = true;
```

### Screenshots

[image]

### Aditional context

OS : Pop!_OS 21.10
Mono : 6.12.0 Preview (6.12.0.147)
.NET Runtime : 6.0 (6.0.3-1) 

VSCode : 1.66.1
.NET nanoFramework extension : 1.0.56
C# extension : 1.24.3

josesimoes, 2022-04-18:
Interesting that this hasn't come up before... 🤔 

For reference: what is the USB serial chip that you have there? And the driver version?

And looking at this, as we're not using DTR, DSR, shouldn't those be set to `false`, instead of `true`?

shodan8192, 2022-04-19:
My USB chip is CP2102, but I've checked also CH341 and FT232RL, all on 5.16.15 kernel - behavior is the same.

DtrEnable and RtsEnable are `false` by default.

Here's what I think is happening :
On Linux serial port open raises RTS and DTR - it's by design. If we look at Mono's SerialPort.cs :
``` cs
bool dtr_enable = false;
bool rts_enable = false;
```
Next, in SerialPortStream contructor we have :
``` cs
public SerialPortStream (string portName, int baudRate, int dataBits, Parity parity, StopBits stopBits,
		bool dtrEnable, bool rtsEnable, Handshake handshake, int readTimeout, int writeTimeout,
		int readBufferSize, int writeBufferSize)
{
	fd = open_serial (portName);
	if (fd == -1)
		ThrowIOException ();
		
	TryBaudRate (baudRate);
	
	if (!set_attributes (fd, baudRate, parity, dataBits, stopBits, handshake))
		ThrowIOException (); // Probably Win32Exc for compatibility

	read_timeout = readTimeout;
	write_timeout = writeTimeout;
	
	SetSignal (SerialSignal.Dtr, dtrEnable);
	
	if (handshake != Handshake.RequestToSend && 
			handshake != Handshake.RequestToSendXOnXOff)
		SetSignal (SerialSignal.Rts, rtsEnable);
}
```
So, `open_serial` causing RTS/DTR assertion. Then :
- 750us later DTR is deasserted because of `dtrEnable == false`, passed from SerialPort.Open()
- 890us later RTS is deasserted because of `rtsEnable == false`, passed from SerialPort.Open()

That results in 140us long reset signal from auto-program circuit on ESP32.
It looks like above code executing slowly, probably it's interpreted (not JIT compiled yet, or at all), also my testing machine is not recent (i7-7700).

I think that best solution will be to set DTR/RTS to `true` before port open if system is Linux - to maintain it's behavior.

josesimoes, 2022-04-19:
> I think that best solution will be to set DTR/RTS to `true` before port open if system is Linux - to maintain it's behavior.

Agreed. Can you submit a PR with this?

shodan8192, 2022-04-19:
Yes - it will be PR for this and #1017

josesimoes, 2022-04-19:
@shodan8192 awesome! We prefer to keep changes atomic, so please make it two PRs. One fixing each issue. 😉

shodan8192, 2022-04-19:
Ok, so I make separated PRs