nanoframework/Home #1019: nanoff 2.0.44 unable to detect esp32-s2 boards

tomglover24, 2022-04-16:
### Tool

nanoff

### Description

Unable to detect Ada Fruit Feather S2 Board using nanoff 2.0.44 (Latest).

``` shell
 ➜ nanoff --target FEATHER_S2 --serialport COM4 --update -v diag
.NET nanoFramework Firmware Flasher v2.0.44+15c350337c
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

Using COM4 @ 1500000 baud to connect to ESP32.
Reading details from chip...

Executing esptool with the following parameters:
'--port COM4  --chip auto   --after no_reset_stub flash_id'

Error E4000: Error executing esptool command. (esptool.py v3.2
Serial port COM4
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...

Traceback (most recent call last):
  File "esptool.py", line 377, in detect_chip
struct.error: unpack requires a buffer of 16 bytes

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "esptool.py", line 5143, in <module>
  File "esptool.py", line 5136, in _main
  File "esptool.py", line 4541, in main
  File "esptool.py", line 121, in get_default_connected_device
  File "esptool.py", line 388, in detect_chip
  File "esptool.py", line 663, in connect
  File "esptool.py", line 616, in _connect_attempt
  File "serial\serialutil.py", line 594, in inWaiting
  File "serial\serialwin32.py", line 259, in in_waiting
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
[8124] Failed to execute script 'esptool' due to unhandled exception!
)
 ```
 
 This seems to be related to version 3.2 of esptool which is included. 
 These are the two related tickets on their tool:
 https://github.com/espressif/esptool/issues/721
 https://github.com/espressif/esptool/issues/685
 
 esptool v3.3-dev seems to be fixed.
 
 Downgrading to nanoff 2.0.15 which was the last release before esptool was upgraded to 3.2 also works. 
 
 2.0.15 output:
 ``` shell
  dotnet tool install -g nanoff --version 2.0.15
You can invoke the tool using the following command: nanoff
Tool 'nanoff' (version '2.0.15') was successfully installed.
  15:03:49  tools  ﮫ 7.214s 
➜ nanoff --target FEATHER_S2 --serialport COM4 --update -v diag
.NET nanoFramework Firmware Flasher v2.0.15+b744f75e4f
Copyright (C) 2019 .NET Foundation and nanoFramework project contributors

** There is a new version available, update is recommended **
** You should consider updating via the 'dotnet tool update -g nanoff' command **
** If you have it installed on a specific path please check the instructions here: https://git.io/JiU0C **

Using COM4 @ 1500000 baud to connect to ESP32.
Reading details from chip...
OK

Connected to:
ESP32-S2 (ESP32-S2FH32)
Features WiFi, Embedded 4MB Flash, 105C temp rating, ADC and temperature sensor calibration in BLK2 of efuse
Flash size 4MB unknown from ST (manufacturer 0x32 device 0x16406)
PSRAM: available
Crystal 40MHz
MAC 7C:DF:A1:96:00:F0

Trying to find FEATHER_S2 in stable repository...OK
Downloading firmware package...OK
Extracting FEATHER_S2-1.8.0.5.zip...OK

Updating to 1.8.0.5

Erasing flash...
OK
Flashing firmware...
Wrote 19008 bytes (11980 compressed) at 0x00001000 in 0.3 seconds (effective 545.9 kbit/s)...
Hash of data verified.
Wrote 980704 bytes (641218 compressed) at 0x00010000 in 11.3 seconds (effective 692.3 kbit/s)...
Hash of data verified.
Wrote 3072 bytes (136 compressed) at 0x00008000 in 0.0 seconds (effective 513.4 kbit/s)...
Hash of data verified.
OK

**********************************************
The connected device is in 'download mode'.
Please reset the chip manually to run nanoCLR.
**********************************************

 ```

### How to reproduce

1. Install Latest Tool following getting started guide
2. Attempt to load firmware onto esp32-s2 based board
3. See Connection Error

Currently Working:
1. Install 2.0.15 version of tool
2. Attempt to load firmware on to esp32-s2 based board
3. Normal output, and board is flashed.

### Expected behaviour

To be able to load firmware onto esp32-s2 boards.

alberk8, 2022-04-18:
Please update to the latest nanoff tool.   The latest nanoff version is   `nanoff 2.0.44+15c350337c `

`dotnet tool update -g nanoff`

josesimoes, 2022-04-18:
@tomglover24 thanks for pointing out that there's a new release of esptool. We'll update the one in nanoff and let's see if this fixes it.

closed 2022-04-18 by josesimoes