nanoframework/Home #1020: Internal error: call to memory allocation during garbage collection!!!

dazzalogan, 2022-04-16:
### Library/API/IoT binding

Unknown

### Visual Studio version

VS2022 Version 17.1.1

### .NET nanoFramework extension version

2022.2.0.8

### Target name(s)

ESP32-Wrover-Dev from FREENOVE

### Firmware version

Unsure

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Mar 25 2022
  Type:        MinSizeRel build, chip rev. >= 0, support for PSRAM
  CLR Version: 1.7.4.92
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  NFAppMemFault, 1.0.0.0
  System.Device.Gpio, 1.0.4.3
  nanoFramework.System.Collections, 1.4.0.3
  nanoFramework.Hardware.Esp32, 1.3.6.3
  System.IO.FileSystem, 1.0.3.3
  nanoFramework.Runtime.Events, 1.10.0.3
  Windows.Storage, 1.4.7.3
  nanoFramework.System.Text, 1.1.3.3
  System.Net, 1.8.1.3
  nanoFramework.Networking.Sntp, 1.5.8.3
  nanoFramework.M2Mqtt, 5.0.2.2
  System.IO.Streams, 1.0.0.2
  nanoFramework.Runtime.Native, 1.5.4.3
  System.IO.Ports, 1.0.4.3
  Windows.Devices.Adc, 1.5.3.2
  nanoFramework.Json, 2.1.2.3
  System.Threading, 1.0.4.3
  Windows.Storage.Streams, 1.12.4.3
  mscorlib, 1.12.0.4
  System.Device.WiFi, 1.4.0.3

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.1, checksum 0xA01012C3
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.4.0, checksum 0xCB7C0ECA
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.1.0, checksum 0x3F6E2A7E
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3f80086c  0x003b0000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

After initial hardware setup, the app. enters main loop in program.cs but then immediately hangs.

Possibly something to do with COM2 being initialised.

PLEASE NOTE: I am new to NF and ESP32, therefore it could be something I'm doing wrong.

### How to reproduce

Just run the app. It should hang as soon as it enters main loop in program.cs.

### Expected behaviour

Continue forever in main loop.

### Sample project or code

[NF Mem Fault.zip](https://github.com/nanoframework/Home/files/8500746/NF.Mem.Fault.zip)

Ellerbach, 2022-04-17:
Thanks for posting the full code, few questions:

- Why do you for the garbage collector?
 
[image]

- Why do you write low and high the pin led? (for debug reasons?)
- For averaging, this is not the correct pattern:
[image]
You should use the pattern with the double and incremental calculation, you will avoid getting an overflow on your int.

- with your serial port, trying to read on 2 different threads will create you issues:

[image]
So use either one either the other. If you need both then use a lock pattern when reading. Serial port is a single thread on purpose, like any hardware. So make sure you're not accessing from 2 different threads like in your case here. So in short, you'll get issues as it's really single thread.

PNPTransistor, 2022-04-17:
Hi Laurent,

Thanks for your reply.
In answer to your questions:

1/ The GC debug was suggested to me at the start because another member thought I might have a memory leak, and that this would capture it.

2/ Yes, the LED is blinking to clearly show that the main loop is running.
NOTE: This app. is purely experimental. It's my first time proper use of NF and I'm just seeing what I can do with it.

3/ I believe the averaging works fine. At least, VS / NF is not reporting any overflows or any other errors to suggest otherwise.
One would hope that if there was an error, VS / NF would inform me?
I will try changing it, though.

4/ Regarding the serial port.
Yes, I admit the code is not ideal here.
However, there's a reason for it being like it is as follows:

I don't fully understand the SerialDataReceiveEvent. I know this event is NOT guaranteed to fire on each character in a desktop environment and I understand this shortfall has worked it's way into NF?
I find the documentation on it confusing. I'm not really sure what it's supposed to do, or how reliable it is, but I also can't comprehend how fast the ESP32 / NF is acting on it, so I'm guessing and experimenting with it at the moment.
Ultimately I want to implement Modbus RTU.
Initially, one of you guys suggested I didn't use this event, but rather just keep reading the quantity of bytes in a loop.
So I'm confused as to what I should do in this respect.

Thanks

alberk8, 2022-04-18:
> Thanks for posting the full code, few questions:
>
> * Why do you for the garbage collector?
>
> [image]
>
> * Why do you write low and high the pin led? (for debug reasons?)
> * For averaging, this is not the [...]

The command is not for garbage collection when a false parameter is passed,  as far as I know it is to get the available nF memory without running the garbage collection.  The naming of it is up for debate, [GC.Run(false)](https://docs.nanoframework.net/api/nanoFramework.Runtime.Native.GC.html).

josesimoes, 2022-04-18:
Any chance you can provide an URL for the solution? I can't download and use ZIPs for security reasons.

alberk8, 2022-04-19:
I have been testing the following configuration that seems to resolve the hang when GC.Run(false) is called.

**app_main.c** (swap the core between receiver_task and main_task)
```
// start receiver task pinned to core 1
 xTaskCreatePinnedToCore(&receiver_task, "ReceiverThread", 3072, NULL, 5, NULL, 0);

 // start the CLR main task pinned to core 0
 xTaskCreatePinnedToCore(&main_task, "main_task", 15000, NULL, 5, NULL, 1);
```

**sdkconfig.default.xxxx**  (add the following)
`CONFIG_ESP32_WIFI_TASK_PINNED_TO_CORE_1=y`

The added benefit of this is, is that the VS debug is a lot more stable.  eg prior to config change,  app deployment completes but the debug session failed, only on subsequent deploy does goes through.  
  
@josesimoes, if you have time please test the new configuration.

closed 2022-05-09 by josesimoes