nanoframework/Home #1022: Project files porperties exeception

Bengkel, 2022-04-18:
### Tool

Visual Studio extension

### Description

I created a nanoframework applications. When I try to access the properties of the project file an exception is thrown.

### How to reproduce

1. Create new project
2. Open project properties

### Screenshots

[image]

### Aditional context

Visual Studio 2022
nanoFramework Extension 2022.1.0.25

Ellerbach, 2022-06-08:
Few more information to help in the work.
The log error shows this:

```text
08/06/2022 08:43:50
LimitedFunctionality
System.AggregateException: Project system data flow 'SupportedTargetFrameworksEnumProvider: 23275750' closed because of an exception.
The given key 'TargetFrameworkIdentifier' was not present in the dictionary. ---> System.Collections.Generic.KeyNotFoundException: The given key 'TargetFrameworkIdentifier' was not present in the dictionary.
   at System.Collections.Immutable.ImmutableDictionary`2.get_Item(TKey key)
   at Microsoft.VisualStudio.ProjectSystem.VS.Frameworks.SupportedTargetFrameworksEnumProvider.Transform(IProjectSubscriptionUpdate input)
   at Microsoft.VisualStudio.ProjectSystem.CommonProjectSystemTools.Derive[T,TOut](IProjectVersionedValue`1 projectVersionedValue, Func`2 transform)
   at Microsoft.VisualStudio.ProjectSystem.TransformBlockSlim`2.TransformBlockSlimSync.ProcessInputAsync(TInput input)
   at Microsoft.VisualStudio.ProjectSystem.DataReceivingBlockSlim`1.<ProcessInputQueueAsync>d__5.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at Microsoft.VisualStudio.ProjectSystem.CommonProjectSystemTools.Rethrow(Exception ex)
   at Microsoft.VisualStudio.ProjectSystem.ExceptionFilter.<>c__DisplayClass2_0.<Guard>g__Action|0()
   at GuardMethodClass.GuardMethod(Func`1 , Func`2 , Func`2 )
   --- End of inner exception stack trace ---
---> (Inner Exception #0) System.Collections.Generic.KeyNotFoundException: The given key 'TargetFrameworkIdentifier' was not present in the dictionary.
   at System.Collections.Immutable.ImmutableDictionary`2.get_Item(TKey key)
   at Microsoft.VisualStudio.ProjectSystem.VS.Frameworks.SupportedTargetFrameworksEnumProvider.Transform(IProjectSubscriptionUpdate input)
   at Microsoft.VisualStudio.ProjectSystem.CommonProjectSystemTools.Derive[T,TOut](IProjectVersionedValue`1 projectVersionedValue, Func`2 transform)
   at Microsoft.VisualStudio.ProjectSystem.TransformBlockSlim`2.TransformBlockSlimSync.ProcessInputAsync(TInput input)
   at Microsoft.VisualStudio.ProjectSystem.DataReceivingBlockSlim`1.<ProcessInputQueueAsync>d__5.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at Microsoft.VisualStudio.ProjectSystem.CommonProjectSystemTools.Rethrow(Exception ex)
   at Microsoft.VisualStudio.ProjectSystem.ExceptionFilter.<>c__DisplayClass2_0.<Guard>g__Action|0()
   at GuardMethodClass.GuardMethod(Func`1 , Func`2 , Func`2 )<---
```

From the NFProjectSystem.CSharp.targets, there is this value:
[image]

So most likely there is something inconsistent here while .NET nanoFramework is not *yet* a proper SDK.

josesimoes, 2022-06-08:
WOW! Nice finding. If that's what happening then we should look into [this method](https://github.com/dotnet/project-system/blob/e9b5eabae85d21e491b19f463bd4993cf73d89f1/src/Microsoft.VisualStudio.ProjectSystem.Managed.VS/ProjectSystem/VS/Frameworks/SupportedTargetFrameworksEnumProvider.cs#L25).

As there is code there that is meant to handle situations where the moniker is not know, we need to figure out if we're providing a wrong or misleading property that's causing that exception.