nanoframework/Home #1023: I2C 2 not properly working on ESP32

Ellerbach, 2022-04-18:
### Library/API/IoT binding

System.Device.I2c

### Target name(s)

ESP32 REV 0 without PSRAM

### Firmware version

1.8.0.6

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Apr  1 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.8.0.6
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  Test2I2cSameTime, 1.0.0.0
  System.Device.Model, 1.1.0.0
  Iot.Device.Am2320, 0.0.0.0
  UnitsNet.RelativeHumidity, 4.128.0.0
  mscorlib, 1.12.0.4
  System.Device.I2c, 1.0.3.3
  nanoFramework.Runtime.Events, 1.10.0.3
  UnitsNet.Temperature, 4.128.0.0
  System.Buffers.Binary.BinaryPrimitives, 1.1.0.0
  nanoFramework.Hardware.Esp32, 1.3.6.3

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.1, checksum 0xA01012C3
  Windows.Devices.Adc v100.1.3.3, checksum 0xCA03579A
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  Windows.Devices.Gpio v100.1.2.2, checksum 0xC41539BE
  Windows.Devices.I2c v100.2.0.2, checksum 0x79EDBF71
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  Windows.Devices.Pwm v100.1.3.3, checksum 0xBA2E2251
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  Windows.Devices.SerialCommunication v100.1.1.2, checksum 0x34BAF06E
  System.IO.Ports v100.1.4.0, checksum 0xCB7C0ECA
  Windows.Devices.Spi v100.1.4.2, checksum 0x360239F1
  System.Device.Spi v100.1.1.0, checksum 0x3F6E2A7E
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

Trying to use both I2C busses on ESP32. Whatever setup is done with the pins, for the exact same sensor connected on both busses, only the one connected on I2C1 will work. The one on I2C2 will never work.
Simple code:

```csharp
            Debug.WriteLine("Hello from nanoFramework!");
            Configuration.SetPinFunction(12, DeviceFunction.I2C1_DATA);
            Configuration.SetPinFunction(14, DeviceFunction.I2C1_CLOCK);
            Am2320 _dhtIn = new(new I2cDevice(new I2cConnectionSettings(1, Am2320.DefaultI2cAddress)));
            Configuration.SetPinFunction(21, DeviceFunction.I2C2_DATA);
            Configuration.SetPinFunction(22,DeviceFunction.I2C2_CLOCK);
            Am2320 _dhtOut = new(new I2cDevice(new I2cConnectionSettings(2, Am2320.DefaultI2cAddress)));

            while(true)
            {
                var tempIn = _dhtIn.Temperature;
                var humIn = _dhtIn.Humidity;
                var tempOut = _dhtOut.Temperature;
                var humOut = _dhtOut.Humidity;
                // Both should be true!
                Debug.WriteLine($"In: {_dhtIn.IsLastReadSuccessful}, out: {_dhtOut.IsLastReadSuccessful}");
                Thread.Sleep(5000);
            }
```

What has been tried:
- changing the pins from I2C1 to I2C2. All pins work fine even when switched.
- changing the sensors, both works fine.

### How to reproduce

Run the previous code with 2 identical sensors on each I2C bus

### Expected behaviour

Having a working I2C2 bus

### Sample project or code

See in description

josesimoes, 2022-04-19:
What's the module (hardware) you're using?

Ellerbach, 2022-04-20:
> What's the module (hardware) you're using?

As per the sample provided:

> ```cs
> Am2320
> ```

Ellerbach, 2022-04-20:
I have about 10 of them, so I've changed them few times, I swap in the code the I2C1 and I2C2 pins, whatever the combination is with 2 proven to work sensors, I always get only the one on I2C1 working, never the one on I2C2

josesimoes, 2022-04-20:
Apologies for not being clear: I was asking about the ESP32 module 😁

Ellerbach, 2022-04-20:
> Apologies for not being clear: I was asking about the ESP32 module 😁

ESP32_REV0 is the image :-) The module is a basic ESP32 without PSRAM, the very classical ones.

Ellerbach, 2022-04-20:
So it now seems to work fine with the default pins. Will check with different pins

closed 2022-04-20 by Ellerbach