nanoframework/Home #1026: Dev container for ESP for latest IDF 4.4.1

alberk8, 2022-04-21:
### Target name(s)

ESP32_PSRAM_REV0, ESP32_REV0

### Firmware version

1.8.0.16 (PSRAM) ,  1.8.0.37 

### Was working before? On which version?

1.8.0.6 (no psram)

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4.1
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Apr 19 2022
  Type:        MinSizeRel build, chip rev. >= 0, support for PSRAM
  CLR Version: 1.8.0.16
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  nanoFramework_Http, 1.0.0.0
  nanoFramework.Runtime.Events, 1.10.0.3
  nanoFramework.Hardware.Esp32, 1.3.6.3
  System.Net, 1.8.1.3
  System.Threading, 1.0.4.3
  nanoFramework.System.Text, 1.1.3.3
  nanoFramework.System.Collections, 1.4.0.3
  nanoFramework.ResourceManager, 1.1.4.3
  System.Net.Http, 1.4.0.0
  System.Device.WiFi, 1.4.0.3
  System.IO.Streams, 1.0.0.2
  mscorlib, 1.12.0.4

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.1, checksum 0xA01012C3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.4.0, checksum 0xCB7C0ECA
  System.Device.Spi v100.1.2.0, checksum 0xB6C4B3BD
  System.Device.WiFi v100.0.6.3, checksum 0x5C9E06C4
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3f80086c  0x003b0000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

System panic when resolving TLS. 

### How to reproduce

1. [HttpRequest - enable the https option ](https://github.com/alberk8/nanoFramework-HttpWebRequest/blob/master/nanoFramework-HttpWebRequest)

### Expected behaviour

Works as expected.

### Screenshots

▒▒▒▒NFPKTV1▒▒ connect... 0
connect... 0
global-root-ca.chain-demos.digicert.com
/
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400940bb  PS      : 0x00060933  A0      : 0x800946ad  A1      : 0x3ffcdbc0
A2      : 0x3ffbc39c  A3      : 0x00000000  A4      : 0x3fbac714  A5      : 0x0000011e
A6      : 0x3fbac180  A7      : 0x00000028  A8      : 0x0000000e  A9      : 0x3ffdf474
A10     : 0x00000005  A11     : 0x000000e0  A12     : 0x3ffbc654  A13     : 0x00000000
A14     : 0x00060923  A15     : 0x00060920  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0000000c  LBEG    : 0x4008bcee  LEND    : 0x4008bcf9  LCOUNT  : 0x00000000

Backtrace:0x400940b8:0x3ffcdbc00x400946aa:0x3ffcdbe0 0x400853e6:0x3ffcdc00 0x400853f9:0x3ffcdc20 0x400f6d63:0x3ffcdc40 0x400f35ad:0x3ffcdc60 0x400f303b:0x3ffcdca0 0x400ec0eb:0x3ffcdcf0 0x400ec1bb:0x3ffcdd40 0x400d9f73:0x3ffcdd60 0x400da041:0x3ffcdd80 0x400d80b4:0x3ffcdda0 0x400d8d41:0x3ffcddc0 0x400e31f5:0x3ffcde00 0x400f7f7e:0x3ffcde30 0x400f6636:0x3ffcde50

ELF file SHA256: c971a79344942de6

Rebooting...

alberk8, 2022-04-22:
I have track it down to this line in native [ssl_generic_init_internal.cpp](https://github.com/nanoframework/nf-interpreter/blob/c07dc5da0c10346cb58c64b8e9e5813217888b92/src/PAL/COM/sockets/ssl/mbedTLS/ssl_generic_init_internal.cpp#L103) that cause the panic.  Not sure what is happening,  it is just a memset.

alberk8, 2022-04-22:
The latest finding I can find is the size mismatch in the `mbedtls_ssl_context` .  I could not find where is the include that is causing the problem or where is the two version coming from. 

**file ssl_generic_init_internal.cpp**
```
  ets_printf("ssl mbedtls_ssl_context size %d\n", sizeof(mbedtls_ssl_context));
  mbedtls_ssl_init(context->ssl);
```
ssl mbedtls_ssl_context size **288**

**sources/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c**
void mbedtls_ssl_init( mbedtls_ssl_context *ssl )
{
    mbedtls_printf("mbedtls_ssl_context size %d\n",(int) sizeof(mbedtls_ssl_context));
    memset( ssl, 0, sizeof( mbedtls_ssl_context ) );
}
mbedtls_ssl_context size **544**

josesimoes, 2022-04-22:
@alberk8 that's valuable information! Thanks for it.

PNPTransistor, 2022-05-03:
Hi there,

How's this issue going?

Sorry, I'm not great with Github and so I don't know where to look for these notifications, etc.

Thanks

josesimoes, 2022-05-04:
@PNPTransistor you can subscribe the issue. That's above. Top right menu at the issue. There's a button with a suggestive name there😉

alberk8, 2022-05-09:
Upon further investigation and testing, this is what I found that with the following changes, the TLS now works.  I am very sure this is not the correct way to a do a proper fix.   As a test it works.   

The actual setting is already set in [FindmbedTLS.cmake](https://github.com/nanoframework/nf-interpreter/blob/main/CMake/Modules/FindmbedTLS.cmake) but not used.   I have deleted the file and even introduce error in it but the build went through without failing.

_Temporary Fix_
Add the following [binutils.ESP32.cmake](https://github.com/nanoframework/nf-interpreter/blob/bb0c260c47a4d3dd22b03d727d4136b0d1d66f47/CMake/binutils.ESP32.cmake#L161)
```
     if(USE_NETWORKING_OPTION)

        nf_add_lib_network(
            BUILD_TARGET
                ${target}
            EXTRA_SOURCES 
                ${TARGET_ESP32_IDF_NETWORK_SOURCES}
            EXTRA_INCLUDES 
                ${ESP32_IDF_INCLUDE_DIRS}
                ${TARGET_ESP32_IDF_INCLUDES}
                ${esp32_idf_SOURCE_DIR}/components/mbedtls/mbedtls/include
                ${esp32_idf_SOURCE_DIR}/components/mbedtls/port/include
                ${esp32_idf_SOURCE_DIR}/components/mbedtls/port/include/mbedtls
        )
```

[mbedtls.h](https://github.com/nanoframework/nf-interpreter/blob/bb0c260c47a4d3dd22b03d727d4136b0d1d66f47/src/PAL/COM/sockets/ssl/mbedTLS/mbedtls.h#L11)
```
#ifdef PLATFORM_ESP32
#include <esp32_idf.h>
#include "esp_config.h"
#else
#include "nf_mbedtls_config.h"
#endif
```

Run Build.

josesimoes, 2022-05-09:
I started work on this a couple of days ago to validate the findings you've reported above.

The reason for the structs having different sizes it's a valid reason for the hang as it would be reading&writing from memory that are not meant to.

Now, the only reason for the mbedtls_context to have different sizes is that the config and defines are coming from different places on the various places the header is used. I recall validating this when doing the rework on IDF 4.4.
That's aligned to your proposed fixed above which could be bringing all the required config and defines to the build.

I'll keep working on this, as it's more than clear that's the root cause of the issue.
Thanks again for the continued effort on finding this! 💯 👏🏻