nanoframework/Home #1030: Add MethodInfo.MakeGenericMethod() and Type[].Where() or IEnumerable<Type>.Where()

KasperJSdeVries, 2022-04-28:
### Description

### MakeGenericMethod
I need to call a generic function with a type decided at runtime, the only way to do this in c# is to use MakeGenericMethod with a type.

### *.Where()
I often need a list of all methods that implement a certain interface and to do this it is very convenient to use .where instead of looping through all of the classes manually.

### How to solve the problem

### MakeGenericMethod()
MakeGenericMethod needs an implementation.

### *.Where()
the IEnumerable interface needs a Where method

### Describe alternatives you've considered

### MakeGenericMethod
I believe that there are no alternatives to this function in the c# standard library

### *.Where()
You can manually loop through all the items in the list and filter them that way.

### Aditional context

example MakeGenericMethod use:
```c#
MethodInfo method = typeof(Sample).GetMethod(nameof(Sample.GenericMethod));
MethodInfo generic = method.MakeGenericMethod(myType);
generic.Invoke(this, null);
```

example *.where use:
```c#
var typesImplementingInterface = typeof(Program).Assembly.Types.GetEnumerable.Where(t => t.IsSubClassOf(IexampleInterface));
```

Ellerbach, 2022-04-28:
Moving this to a discussion as it's not a bug.
Few remarks: there is not *yet* generics support in .NET nanoFramework.
And there is no Linq support.
Those are very small and constraints environment. Choices need to be done to fit into a very small footprint and memory.
What you describe seems to be able to be achieved with a foreach loop on the methods.

KasperJSdeVries, 2022-04-28:
Thank you for the feedback. I will try other ways to achieve the same functionality.

Ellerbach, 2022-04-28:
Check the Unit Test Launcher: https://github.com/nanoframework/nanoFramework.TestFramework/blob/871c6a0494cecfc53f0ead1775a402f0363afb02/source/UnitTestLauncher/Program.cs#L25

It uses attributes to find functions. Depending on what you need, it can be a good way.

Check out as well the json class, there are a lot of boxing and un boxing:
https://github.com/nanoframework/nanoFramework.Json/blob/382300aaa16c05c2f9d6415dbf97539ce08a703c/nanoFramework.Json/JsonConvert.cs#L242

it checks simple types, it checks arrays and hashtable, it check objects.