nanoframework/Home #1035: No SerialPort.WatchChar events if WatchChar set before SerialPort opened

KiwiBryn, 2022-05-06:
### Library/API/IoT binding

System.IO.Ports

### Visual Studio version

VS2019

### .NET nanoFramework extension version

2019.11.0.13

### Target name(s)

ST_STM32F769I_DISCOVERY

### Firmware version

1.8.0.58

### Device capabilities

System Information
HAL build info: nanoCLR running @ ST_STM32F769I_DISCOVERY
  Target:   ST_STM32F769I_DISCOVERY
  Platform: STM32F7

Firmware build Info:
  Date:        May  3 2022
  Type:        MinSizeRel build with ChibiOS v2021.11.1.11
  CLR Version: 1.8.0.58
  Compiler:    GNU ARM GCC v10.3.1

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: YES
  nanoBooter: v1.8.0.58
  IFU capable: NO
  Has proprietary bootloader: NO

AppDomains:

Assemblies:
  BreakOutSerial, 1.0.0.0
  System.IO.Ports, 1.0.4.3
  mscorlib, 1.12.0.4
  nanoFramework.Runtime.Events, 1.10.0.3
  nanoFramework.System.Text, 1.1.3.3
  System.IO.Streams, 1.0.0.2

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Stm32 v100.0.4.4, checksum 0x0874B6FE
  nanoFramework.Devices.Can v100.0.5.0, checksum 0xF319D40E
  nanoFramework.Graphics v100.0.0.3, checksum 0xB3EDAF22
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.1, checksum 0xA01012C3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.4.0, checksum 0xCB7C0ECA
  System.Device.Spi v100.1.2.0, checksum 0xB6C4B3BD
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0xc0000000  0x01000000
  FLASH 0x08000000  0x00200000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x08000000       1      0x008000     nanoBooter
      1    0x08008000       1      0x008000     Configuration
      2    0x08010000       2      0x008000     nanoCLR
      3    0x08020000       1      0x020000     nanoCLR
      4    0x08040000       2      0x040000     nanoCLR
      5    0x080C0000       5      0x040000     Deployment

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x08000000    0x008000 (32kB)     nanoBooter
  0x08008000    0x008000 (32kB)     Configuration
  0x08010000    0x0B0000 (704kB)    nanoCLR
  0x080C0000    0x140000 (1280kB)   Deployment

### Description

If the SerialPort WatchChar is set before opening the serial port no SerialData.WatchChar events. 

Also noticed that SerialData.WatchChar tooltip maybe incorrect.

### Expected behaviour

There should SerialData.WatchChar events if I set SerialPort.WatchChar before opening the port.

### Screenshots

[image]

### Sample project or code

I used the SerialCommunication sample

Requires small mods to Scenario1_ConfigureDevice.cs & Scenario4_DataReceivedEvent.cs to repro. I used a serial device that responded to Hayes AT Commands, but a simple serial loopback (TX pin connected to RX pin on device) should work 

Around line 30 in Scenario4_DataReceivedEvent.cs 
            serialDevice.NewLine = "\r\n";

            serialDevice.WriteLine("at+version"); // Added this line to send short message

            // set a watch char to be notified when it's available in the input stream
            //serialDevice.WatchChar = '\n';  // Commented out this line 

            Debug.WriteLine("Waiting to receive data in serial port...");

Around line 72 in Scenario1_ConfigureDevice.cs
          // if dealing with massive data input, increase the buffer size
            _serialDevice.ReadBufferSize = 2048;

            _serialDevice.WatchChar = '\n'; // Before open WatchChar event doesn't fire

            // open the serial port with the above settings
            _serialDevice.Open();

            _serialDevice.WatchChar = '\n'; // After Open WatchChar event fires

josesimoes, 2022-05-06:
Confirmed!

networkfusion, 2022-05-11:
🤔 I wonder if it is as simple as removing `if (_opened)` from https://github.com/nanoframework/System.IO.Ports/blob/fa420c623128cbc1010bf467374653469212071c/System.IO.Ports/SerialPort.cs#L341
[image]

josesimoes, 2022-05-11:
Unfortunately no... It's a bit deeper and wider than than.

networkfusion, 2022-05-11:
In that case, I will just add a PR for @KiwiBryn 's comment about incorrect intelisense
https://github.com/nanoframework/System.IO.Ports/pull/83

networkfusion, 2022-05-11:
This might also be of interest (although technically unrelated to this particular issue)... https://github.com/nanoframework/System.IO.Ports/pull/84

closed 2022-05-26 by josesimoes