nanoframework/Home #1036: Gpio Read on ESP32 always return low (zero)

alberk8, 2022-05-09:
### Target name(s)

ESP32_REV0

### Firmware version

1.8.0.120

### Was working before? On which version?

N/A

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4.1
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        May  8 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.8.0.120
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x9A53BB44
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.4.1, checksum 0xA01012C3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.4.0, checksum 0xCB7C0ECA
  System.Device.Spi v100.1.2.0, checksum 0xB6C4B3BD
  System.Device.Wifi v100.0.6.4, checksum 0x1C1D3214
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

GpioPin.Read() always return zero.

### How to reproduce

```
private static GpioController _gpio;
private static GpioPin _led;
public static void Main()
{
  _gpio = new GpioController();
  _led = _gpio.OpenPin(21, PinMode.OutputOpenDrainPullUp);
  while (true)
    {
       Console.WriteLine("Led: " + _led.Read().ToString());
       if (_led.Read() == PinValue.High)
       {
          _led.Write(PinValue.Low);
       }
       else
       {
          _led.Write(PinValue.High);
       }

       Thread.Sleep(100);
     }
}
```

### Expected behaviour

Able to read the GpioPin value.

### Aditional information

Add the following code in [CPU_GPIO_SetDriveMode](https://github.com/nanoframework/nf-interpreter/blob/bb0c260c47a4d3dd22b03d727d4136b0d1d66f47/targets/ESP32/_nanoCLR/System.Device.Gpio/cpu_gpio.cpp#L530)  This seems to fix the Gpio Output but have not tested with input as well as all the code that depends on  'CPU_GPIO_SetDriveMode'    

```
    gpio_config(&GPIOConfig);

    switch(mode)
    {
        case GPIO_MODE_OUTPUT:
            gpio_set_direction((gpio_num_t)pinNumber, GPIO_MODE_INPUT_OUTPUT);
            break;
        case GPIO_MODE_OUTPUT_OD:
            gpio_set_direction((gpio_num_t)pinNumber, GPIO_MODE_INPUT_OUTPUT_OD);
            break;
        default:
            gpio_set_direction((gpio_num_t)pinNumber, mode);
            break;
    }

    return true;
```

josesimoes, 2022-05-09:
Being able to _read_ from a GPIO configured to output was not implemented by design.

On several MCUs is not possible to do this. On other it requires extra steps on configuration (like the above) or extra steps to access specif registers to access that, which lead in extra execution time penalty.
Considering all those mixed scenarios and availability drove that design decision.

It can be argued that an exception could be thrown when performing the Read, but that too was considered a bit harsh.
Both these (not throwing an exception and not being able to actually read from an output gpio) are aligned with .NET IoT behavior.

Just noticed that our documentation about Read is not correct as the description was the one inherited from UWP, which is wrong.

To wrap this up: if an application needs to keep state of the output value, it has to do it on it's own.
For simple operations like toggling a GPIO, we've provided an API for that: `Toggle()`.

closed 2022-05-09 by josesimoes

KiwiBryn, 2022-05-09:
I understand why this is the case but it would be good if .Read not working on some devices was highlighted somewhere. 

Burnt some time yesterday because some code which used .Read worked on my Netduino W3 Wifi, and Disco769 (main test platforms) using but not on my Sparkfun ESP32 Single Channel gateway.

Bryn

josesimoes, 2022-05-09:
@KiwiBryn you correct. Doca should be as complete as possible and behavior across platforms consistent. But we have a very small team lot of work to do and sometimes we fail.

The docs will be improved to better expose this.

Apologies for any inconvenience this may have caused you.

alberk8, 2022-05-10:
I still feel that Gpio.Read should return the correct value regardless of whether we have event trigger to get that value as it changes.   There are a lot of times you just want to read the current value of the GPIO.

josesimoes, 2022-05-10:
@alberk8 you'll get a proper value for the state of the gpio when calling Read. Providing the gpio is configured as Input.