nanoframework/Home #1037: Cannot get an intance of `AdcChannel` on `M5Core2`

timothyparez, 2022-05-09:
### Library/API/IoT binding

nanoFramework.M5Stack

### Target name(s)

M5Core2

### Device capabilities

ESP32 (ESP32-D0WDQ6-V3 (revision 3))
Features WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Flash size 16MB unknown from ST (manufacturer 0x32 device 0x16408)
PSRAM: undetermined
Crystal 40MHz
MAC 30:C6:F7:25:DE:28

### Description

Following the samples in `readme.md` from https://github.com/nanoframework/nanoFramework.M5Stack  
I am trying to read the value of an ADC pin.  

As soon as I try to get an instance of  `AdcChannel` the code stops executing  
(Probably throws an exception but since I'm not using Visual Studio, I cannot see the output)

### How to reproduce

1. Create a project with the following code

```
using System;
using System.Diagnostics;
using System.Threading;
using nanoFramework.M5Core2;
using nanoFramework.M5Stack;
using nanoFramework.M5Stack;
using nanoFramework.Presentation.Media;
using Console = nanoFramework.M5Stack.Console;
using System.Device.Adc;

namespace Example
{
    public class Program
    {
        public static void Main()
        {                        
            M5Core2.InitializeScreen();
            AdcChannel adc = M5Core2.GetAdcGpio(36);

            //The code below is never executed
            //If you remove the line above and set value explicitly it works

            while(true)
            {
                Console.CursorTop = 1;
                Console.CursorLeft = 0;                                
                var value = adc.ReadValue();
                Console.WriteLine(value.ToString());
                Console.WriteLine("-----------------");
                Thread.Sleep(500);
            }
        }
    }
}
```

### Expected behaviour

Expected to be able to read the ADC value

### Sample project or code

https://github.com/timothyparez/nanoframeworkadcissue

### Aditional information

- Ubuntu 22.04
- VSCode
- .NET 6
- Mono 6.12.0.147

timothyparez, 2022-05-09:
Looking at [M5Core2.cs](https://github.com/nanoframework/nanoFramework.M5Stack/blob/280e4dfd6963be758a47c758175866941b6d123e/nanoFramework.M5Core2/M5Core2.cs#L334) I can see only `32`, `39`, `0`, `2`, `4`, `12`, `15`, `25` and `27` are allowed.  
The M5Stack [readme.md](https://github.com/nanoframework/nanoFramework.M5Stack/blob/main/README.md#adc-channels) however shows an example with pin `35`:

-----
ADC Channels are pre setup on the M5Core, M5Core2, Fire and Atom Lite/Matrix, access them like this:

     // This will give you the ADC1 channel 7 which is on pin 35 of M5Core
     AdcChannel myChannel = M5Core.GetAdcGpio(35);
------

On the back of the M5Core2 device the ADC pins are named `G35` and `G36`

networkfusion, 2022-05-11:
There are definitely some inconsistencies there (going by my board)
[image]

@timothyparez Feel free to create a PR.