nanoframework/Home #1041: Serial.IO.Ports can hang all threads

AdrianSoundy, 2022-05-10:
### Target name(s)

ESP32

### Firmware version

ALL

### Was working before? On which version?

no

### Description

If you are sending data on a serial port with hand shaking enabled
If the hand shaking is stopping the send the call to write to serial ports hangs all managed threads until send completes.

This only happens to short messages which don't use long running path. 

As call is stuck in the IDF "uart_write_bytes" call, all managed threads stop.  
Which can cause a dead lock as in my case serial reads are not being done and hand shake restarted.

There is also a performance issues as call always waits for bytes to be sent. I have a fix for this and noticed performance difference.

### Expected behaviour

The calls to serial writes do not stop managed threads running.

AdrianSoundy, 2022-05-10:
The fix is to use a different call which just puts data into available space in FiFo and any remaining data done in long running mode. This seems to work well.

josesimoes, 2022-05-11:
I understand the problem and that makes sense. If it can be solved by means of the IDF API: great. If not we have to add a thread continuation pattern there.

closed 2022-05-13 by josesimoes