nanoframework/Home #1043: Unable to build ESP32 firmware on Linux, due to incorrect source file names casing

shodan8192, 2022-05-12:
### Target name(s)

ESP32

### Description

File names in cmake input files doesn't match case of source file names, which prevent correct building firmware on Linux

- FindSystem.Device.Spi.cmake `nanoHAL_spi.cpp`, actual name `nanoHAL_Spi.cpp`

- FindSystem.Device.Wifi.cmake `sys_dev_wifi_native_System_Device_WiFi_WiFiAdapter.cpp`, actual name `sys_dev_wifi_native_System_Device_Wifi_WifiAdapter.cpp`

josesimoes, 2022-05-12:
Both fixes should be made in the CMakelists.txt files as the file names are correct.

shodan8192, 2022-05-12:
OK, I can make PR for this :smiley:

closed 2022-05-12 by josesimoes