nanoframework/Home #1045: GpioPin opened by SpiDevice is not working

KasperJSdeVries, 2022-05-20:
### Library/API/IoT binding

System.Device.Spi or System.Device.Gpio

### Visual Studio version

VS2022

### Target name(s)

ST_STM32F329I_DISCOVERY

### Device capabilities

System Information
HAL build info: nanoCLR running @ ST_STM32F429I_DISCOVERY
  Target:   ST_STM32F429I_DISCOVERY
  Platform: STM32F4

Firmware build Info:
  Date:        May 14 2022
  Type:        MinSizeRel build with ChibiOS v2021.11.1.11
  CLR Version: 1.8.0.144
  Compiler:    GNU ARM GCC v10.3.1

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: YES
  nanoBooter: v1.8.0.144
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  BlueNrg2.Samples, 1.0.0.0
  mscorlib, 1.12.0.4
  System.Device.Gpio, 1.0.4.3
  nanoFramework.Runtime.Events, 1.10.0.3
  System.Device.Spi, 1.2.1.12878
  BlueNrg2, 1.0.0.0
  nanoFramework.Logging, 1.0.1.13

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Stm32 v100.0.4.4, checksum 0x0874B6FE
  nanoFramework.Device.Can v100.0.5.1, checksum 0x5B569036
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.4.0, checksum 0xCB7C0ECA
  System.Device.Spi v100.1.2.0, checksum 0xB6C4B3BD

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0xd0000000  0x00800000
  FLASH 0x08000000  0x00200000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x08000000       1      0x004000     nanoBooter
      1    0x08004000       3      0x004000     nanoCLR
      2    0x08010000       1      0x010000     nanoCLR
      3    0x08020000       1      0x020000     nanoCLR
      4    0x08040000       6      0x020000     Deployment
      5    0x08100000       4      0x004000     Deployment
      6    0x08110000       1      0x010000     Deployment
      7    0x08120000       7      0x020000     Deployment

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x08000000    0x004000 (16kB)     nanoBooter
  0x08004000    0x03C000 (240kB)    nanoCLR
  0x08040000    0x1C0000 (1792kB)   Deployment

Deployment Map
Empty

### Description

After a pin has been opened by creating an SpiDevice, you can't read the pin.

### How to reproduce

```cs
var cs = Utilities.GetPinNumber('E', 4); // sets cs to 68
var spiDevice = new SpiDevice(new SpiConnectionSettings(4, cs)); // Creates a SpiDevice using the cs pin we set above.
var controller = new GpioController(); // creates a GpioController

Debug.WriteLine($"{controller.GetPinMode(cs)}"); // reads the pin used by the SpiDevice
```
the Writeline throws an invalidOperationError.

### Expected behaviour

it runs without throwing an error, and writes "3" to the stdout

### Screenshots

[image]

networkfusion, 2022-05-20:
I have not tested, but I think you have to open a pin before you can get its `mode`.
`gpioController.OpenPin(Utilities.GetPinNumber('E', 4), PinMode.Output);`
However, if this is the case, we probably need to throw an error to that affect.
Otherwise it is possible that it is an issue upstream (Note: this URL is only an up-to-date fork, which is easier to view than the SVN repo):
https://github.com/ArduPilot/ChibiOS.svn/tree/master/os/hal/boards/ST_STM32F429I_DISCOVERY

josesimoes, 2022-05-20:
Renamed the issue as it was a bit _too_ much dramatic. It sounded like GPIO wasn't working at all.

josesimoes, 2022-05-20:
To @networkfusion point above, that could be the explanation for this. The reserved pins are kept on a structure in native code which is not reflected at C#. And the code in GetPinMode starts by looking at the array in C#. This mismatch is (99% sure) the root cause for this.

KasperJSdeVries, 2022-05-23:
@networkfusion Pin `E2` is the clock pin not the chip-select pin, the chip-select pin should be a separate GPIO.

on another note: @josesimoes why have the chip-select be managed by the SpiDevice? There are applications where the chip-select needs to be toggled for things other than SPI (e.g. resets etc.).

KasperJSdeVries, 2022-05-23:
Some more precise code blocks showing the issue:

### Works:
```cs
var cs = Utilities.GetPinNumber('E', 4);
var controller = new GpioController();

controller.OpenPin(cs);

Debug.WriteLine($"{controller.GetPinMode(cs)}");

Thread.Sleep(Timeout.Infinite);
```

### Throws exception
```cs
var cs = Utilities.GetPinNumber('E', 4);
var controller = new GpioController();

controller.OpenPin(cs);

var spiDevice = new SpiDevice(new SpiConnectionSettings(4, cs)); // <-- Exception thrown here

Debug.WriteLine($"{controller.GetPinMode(cs)}");

Thread.Sleep(Timeout.Infinite);
```

### Throws InvalidOperationException
```cs
var cs = Utilities.GetPinNumber('E', 4);
var controller = new GpioController();

var spiDevice = new SpiDevice(new SpiConnectionSettings(4, cs));

Debug.WriteLine($"{controller.GetPinMode(cs)}"); // <-- Exception thrown here

Thread.Sleep(Timeout.Infinite);
```

### Works
```cs
var cs = Utilities.GetPinNumber('E', 4);
var controller = new GpioController();

var spiDevice = new SpiDevice(new SpiConnectionSettings(4, cs));

// This prints: "BusID: 4, ChipSelectLine: 68"
Debug.WriteLine($"BusID: {spiDevice.ConnectionSettings.BusId}, ChipSelectLine: {spiDevice.ConnectionSettings.ChipSelectLine}");

Thread.Sleep(Timeout.Infinite);
```

As I stated earlier I think the chip-select line should not be managed solely by the SpiDevice class.

josesimoes, 2022-05-23:
> on another note: @josesimoes why have the chip-select be managed by the SpiDevice? There are applications where the chip-select needs to be toggled for things other than SPI (e.g. resets etc.).

@KasperJSdeVries because it's easier for the majority of the use cases. Meaning that you don't have to worry about managing the CS pin. The SPI code will do it for you.
Understood that there are times where you'll want to control the CS pin at your will, because it's hijacking the function that SPI protocol has for it. We are not offering that possibility now. That could be opened for discussion looking into adding it (or not) in a future version.

KasperJSdeVries, 2022-05-23:
> @KasperJSdeVries because it's easier for the majority of the use cases. Meaning that you don't have to worry about managing the CS pin. The SPI code will do it for you.

The chip-select line does not do that much in the SPI protocol, you just have to be careful to not send valid messages while the chip-select is toggled. However some IoT devices may use the chip-select for the entire "module", this means it also needs to get toggled when you want to trigger e.g. reset or other GPIO functionality. 

#### Possible implementation
It should not be too difficult to add an optional parameter to the SpiDevice constructor. For example:
```cs
public SpiDevice(SpiConnectionSettings settings, bool manualChipSelect = false) {...}
```
I don't know how much implementing this impacts the native code.

#### Other quality-of-life functionality
Currently it is not possible to have multiple slaves on a single SpiDevice as you can only specify a single chip-select line. This functionality is fully supported by the protocol.

closed 2022-07-06 by josesimoes