nanoframework/Home #1060: ESP32 - Failed memory check @ 0x001B0000.

IvailoIvanovIvanov, 2022-06-02:
### Library/API/IoT binding

nanoFramework.WebServer

### Visual Studio version

VS2022  17.3.0 Preview 1.

### .NET nanoFramework extension version

2022.2.0.23

### Target name(s)

ESP32-WROOM ESP32-D0WDQ6 (revision 1)

### Firmware version

ESP32_REV0-1.8.0.220

### Device capabilities

ESP32 (ESP32-D0WDQ6 (revision 1))
Features WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Flash size 4MB unknown from  (manufacturer 0x216 device 0x16406)
PSRAM: not available
Crystal 40MHz

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4.1
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        May 27 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.8.0.220
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x0A915860
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
[... 371 lines cut ...]
11:51:57.927 [Deploying 131560/153588 bytes.]
11:51:57.974 [Deploying 132572/153588 bytes.]
11:51:58.022 [Deploying 133584/153588 bytes.]
11:51:58.068 [Deploying 134596/153588 bytes.]
11:51:58.114 [Deploying 135608/153588 bytes.]
11:51:58.160 [Deploying 136620/153588 bytes.]
11:51:58.206 [Deploying 137632/153588 bytes.]
11:51:58.256 [Deploying 138644/153588 bytes.]
11:51:58.302 [Deploying 139656/153588 bytes.]
11:51:58.348 [Deploying 140668/153588 bytes.]
11:51:58.394 [Deploying 141680/153588 bytes.]
11:51:58.440 [Deploying 142692/153588 bytes.]
11:51:58.488 [Deploying 143704/153588 bytes.]
11:51:58.536 [Deploying 144716/153588 bytes.]
11:51:58.585 [Deploying 145728/153588 bytes.]
11:51:58.632 [Deploying 146740/153588 bytes.]
11:51:58.679 [Deploying 147752/153588 bytes.]
11:51:58.726 [Deploying 148764/153588 bytes.]
11:51:58.772 [Deploying 149776/153588 bytes.]
11:51:58.818 [Deploying 150788/153588 bytes.]
11:51:58.866 [Deploying 151800/153588 bytes.]
11:51:58.913 [Deploying 152812/153588 bytes.]
11:51:58.979 [Failed memory check @ 0x001B0000.]
11:51:58.980 [*** ERROR deploying assemblies to the device ***]
11:51:58.979 [*** ERROR: deployment failed ***]

alberk8, 2022-06-03:
Do you have the same issue if you don't use the nanoframework.WebServer?

IvailoIvanovIvanov, 2022-06-03:
> Do you have the same issue if you don't use the nanoframework.WebServer?
If I understood the question correctly... When I run Blinky or GPIO sample works fine.

lsd-techno, 2022-06-03:
Another solution that not includes any WebServer or network packages have same problem when start debug.
I try with another ESP32 board too.
After I get deploy error I ignore it so I can go to debug as seen in the output.
here is some output:

```
01:50:22.500 [Deploying 92092/92336 bytes.]
01:50:22.564 [Failed memory check @ 0x001B0000.]
01:50:22.564 [*** ERROR deploying assemblies to the device ***]
01:50:22.564 [*** ERROR: deployment failed ***]
01:50:31.464 [Launching debugger provider from v2019.11.0.23+ec14f221b2]
01:50:32.018 [Attempting to connect the debugger engine (1/5)]
01:50:32.018 [Assigning debug engine of target device]
01:50:32.019 [Device running nanoCLR, updating the debug flags...]
01:50:32.719 [Attempting to connect the debugger engine (1/5)]
01:50:32.719 [Device running nanoCLR, updating the debug flags...]
01:50:33.328 [Device is running nanoCLR, requesting a restart and pause of debugger (1/5).]
01:50:33.420 [nanoCLR restarted successfully.]
01:50:33.923 [Attempting to connect the debugger engine (1/5)]
01:50:33.923 [Device running nanoCLR, updating the debug flags...]
01:50:34.600 [Device is running the CLR in initialized state]
01:50:34.601 [The debugger engine is listening to debug events from the nanoDevice target]
01:50:35.353 [Updating information about assemblies on the debugging target.]

```

I did try to downgrade to older version both VS2019 extension and nanoFF around 2 month old no changes...

closed 2022-06-06 by josesimoes

josesimoes, 2022-06-07:
> I have still the same issue on this projekt.
> 
> https://github.com/nanoframework/Samples/tree/main/samples/Webserver/WebServer.Sample

@AdrianFishy have you updated the firmware?

dionmills, 2024-04-19:
Im having this issue right now im using all latest firmware and packages

`  <ItemGroup>
    <Reference Include="Iot.Device.Ssd13xx">
      <HintPath>..\packages\nanoFramework.Iot.Device.Ssd13xx.1.3.350\lib\Iot.Device.Ssd13xx.dll</HintPath>
    </Reference>
    <Reference Include="mscorlib">
      <HintPath>..\packages\nanoFramework.CoreLibrary.1.15.5\lib\mscorlib.dll</HintPath>
    </Reference>
    <Reference Include="nanoFramework.Hardware.Esp32">
      <HintPath>..\packages\nanoFramework.Hardware.Esp32.1.6.12\lib\nanoFramework.Hardware.Esp32.dll</HintPath>
    </Reference>
    <Reference Include="nanoFramework.Json">
      <HintPath>..\packages\nanoFramework.Json.2.2.117\lib\nanoFramework.Json.dll</HintPath>
    </Reference>
    <Reference Include="nanoFramework.Runtime.Events">
      <HintPath>..\packages\nanoFramework.Runtime.Events.1.11.15\lib\nanoFramework.Runtime.Events.dll</HintPath>
    </Reference>
    <Reference Include="nanoFramework.System.Collections">
      <HintPath>..\packages\nanoFramework.System.Collections.1.5.31\lib\nanoFramework.System.Collections.dll</HintPath>
    </Reference>
    <Reference Include="nanoFramework.System.Text">
      <HintPath>..\packages\nanoFramework.System.Text.1.2.54\lib\nanoFramework.System.Text.dll</HintPath>
    </Reference>
    <Reference Include="System.Device.Gpio">
      <HintPath>..\packages\nanoFramework.System.Device.Gpio.1.1.38\lib\System.Device.Gpio.dll</HintPath>
    </Reference>
    <Reference Include="System.Device.I2c">
      <HintPath>..\packages\nanoFramework.System.Device.I2c.1.1.16\lib\System.Device.I2c.dll</HintPath>
    </Reference>
    <Reference Include="System.Device.Wifi">
      <HintPath>..\packages\nanoFramework.System.Device.Wifi.1.5.81\lib\System.Device.Wifi.dll</HintPath>
    </Reference>
    <Reference Include="System.IO.Streams">
      <HintPath>..\packages\nanoFramework.System.IO.Streams.1.1.59\lib\System.IO.Streams.dll</HintPath>
    </Reference>
    <Reference Include="System.Math">
      <HintPath>..\packages\nanoFramework.System.Math.1.5.43\lib\System.Math.dll</HintPath>
    </Reference>
    <Reference Include="System.Net">
      <HintPath>..\packages\nanoFramework.System.Net.1.10.77\lib\System.Net.dll</HintPath>
    </Reference>
    <Reference Include="System.Net.Http">
      <HintPath>..\packages\nanoFramework.System.Net.Http.1.5.134\lib\System.Net.Http.dll</HintPath>
    </Reference>
    <Reference Include="System.Threading">
      <HintPath>..\packages\nanoFramework.System.Threading.1.1.32\lib\System.Threading.dll</HintPath>
    </Reference>
  </ItemGroup>`

`
System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF 38eeba2
  Target:   ESP32_REV3
  Platform: ESP32

Firmware build Info:
  Date:        Apr 19 2024
  Type:        MinSizeRel build, chip rev. 3, without support for PSRAM
  CLR Version: 1.9.1.65
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  System.Device.Wifi, 1.5.81.0
  System.Device.Gpio, 1.1.38.0

Native Assemblies:
  mscorlib v100.5.0.19, checksum 0x445C7AF9
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.10.0, checksum 0x6A20A689
  nanoFramework.Hardware.Esp32.Rmt v100.0.4.0, checksum 0x608C5658
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.System.IO.Hashing v100.0.0.1, checksum 0xEBD8ED20
  nanoFramework.System.Security.Cryptography v100.0.0.2, checksum 0xF4AEFE6C
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.1.0.0, checksum 0x62EBC911
  System.Math v100.0.5.5, checksum 0x9F9E2A7E
  System.Net v100.2.0.1, checksum 0xD82C1452
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.6, checksum 0x097E7BC5
  System.Device.I2c v100.0.0.2, checksum 0xFA806D33
  System.Device.I2c.Slave v1.0.0.0, checksum 0x4238164B
  System.Device.I2s v100.0.0.1, checksum 0x478490FE
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.1, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0x3F6E2A7E
  System.Runtime.Serialization v100.0.0.0, checksum 0x0A066871
  System.Device.Wifi v100.0.6.4, checksum 0x00A058C6
  Windows.Storage v100.0.3.0, checksum 0xF0C37E1B`

13:36:05.211 [Deploying 212960/212960 bytes.]
13:36:05.321 [Failed memory check @ 0x001B0000.]
13:36:05.322 [*** ERROR deploying assemblies to the device ***]
13:36:05.321 [*** ERROR: deployment failed ***]

josesimoes, 2024-04-19:
@dionmills what about VS extension? is it the latest one?
Also worth checking if power supply is good on the board, otherwise flash operations are know to fail.