nanoframework/Home #1067: Move to SDK style projects

josesimoes, 2022-06-06:
### Description

patagonaa, 2024-03-21:
Has there been any progress on this lately?
I'm new to nanoFramework and so far this is the only thing that has stuck out as somewhat annoying, as
- transitive dependencies don't work properly (because `packages.config` is used instead of `PackageReference`, at least by default?)
- forgetting to "Save all" before committing means the files are missing in the project file
- projects can't be multi-target easily (as far as I can tell, the best way is two separate projects with the individual files linked from one to the other)
- AssemblyInfo properties (like the version and copyright information) can't be overwritten by `Directory.Build.props`

josesimoes, 2024-03-21:
I hear your pain. Same here.
There are two parts to have this:
1. Migrate project system, which we have pretty much done.
2. VS extensibility SDK has to have support for some features we need related with the above, namely allowing language sub-project type. Hopefully the new extensibility will have all that and then we can close the circle.

patagonaa, 2024-03-21:
Thanks for the update, looking forward to it (and the workaround using links for multi-targeting is good enough for me right now).

Also just wanted to say that nanoFramework is pretty awesome.
I wrote some code for RS485 serial communication for .NET 8 recently and after only minor changes (replacing `Span<byte>` with `SpanByte` (and `using SpanByte = Span<byte>` for full .NET), using synchronous methods instead of `async`, replacing ranges with manual indexing, replacing tuples with classes), it compiled with (almost) no framework-specific code and worked pretty much instantly on an ESP32 just like it does on desktop.
That was my first contact/use of nanoFramework and I would never have guessed it would work so easily (including the pretty much perfect debugging experience).

josesimoes, 2024-03-21:
@patagonaa regarding code sharing, you may want to consider shared projects. That works pretty well too and may be more convenient than linking files.

patagonaa, 2024-03-21:
> regarding code sharing, you may want to consider shared projects. That works pretty well too and may be more convenient than linking files.

looks like I could use that. It would probably mean another extra project (one shared project, one .NET 8 class library, one nanoFramework class library), as (at least for the .NET 8 side) I'd still like to reference an actual library instead of the source code itself, but I don't think that's an issue.
Linking the files works well too though, as surprisingly, the integration in VS is pretty good (switching between both frameworks can be done in the top bar in the source code editor and the hover-tooltip even shows which methods are available in which frameworks, just like it works when using SDK projects with multiple target frameworks)