nanoframework/Home #1071: Add Support for ESP32 C3

josesimoes, 2022-06-07:
### Description

Add support for the C3 series.

KiwiBryn, 2022-10-09:
Got one of these sitting on my desk ready to go for testing...
https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html
[image]

lanmark7, 2022-10-17:
Hoping to use nanoframework with the C3 series..  the seeed board is a really nice configuration of wifi/bluetooth as well as it has a charge controller for adding a rechargeable battery.   Has there been any movement on adding C3 support?

torbacz, 2022-11-28:
@josesimoes Haven't you done it?

josesimoes, 2022-11-28:
@torbacz yes! Just forgot to link to this issue here... 🤦🏻‍♂️

josesimoes, 2022-11-28:
Implemented with nanoframework/nf-interpreter#2471.

closed 2022-11-28 by josesimoes

goblinmaks, 2024-02-14:
ESP32-C3 rev.0.4 not supported 

[image]
@josesimoes is possible to add support ?