nanoframework/Home #1073: Bluetooth Gatt CCCD exponent parameter type/size incorrect

axa88, 2022-06-07:
### Library/API/IoT binding

nanoFramework.Device.Bluetooth

### Target name(s)

ESP32_BLE_REV0

### Device capabilities

ESP32 (ESP32-D0WDQ6 (revision 1))
Features WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Flash size 4MB unknown from ST (manufacturer 0x32 device 0x16406)
PSRAM: not available
Crystal 40MHz
MAC 24:6F:28:10:97:BC

### Description

There is a bug/specification error in the 2904 CCCD implementation.  The exponent parameter should be a `byte`, but nf specifies and sends it, as an `int`

So basically the `CreateGattPresentationFormat` method signature should be changed to:

```
public void CreateGattPresentationFormat(
      byte formatType,
      byte exponent,
      ushort unit,
      byte namespaceId,
      ushort decsription)
```

And thus its payload size changed from 4 to 1 byte for a total payload of 7 bytes not 10

### How to reproduce

create a ble gatt characteristic with the CCCD setting the presentation format to anything...
See the payload has 3 extra bytes for the 2904 characteristic where the unit parameter should begin.

### Expected behaviour

the exponent is only 8 bits not 32

### Screenshots

[image]

josesimoes, 2022-06-07:
cc @AdrianSoundy

alberk8, 2022-06-08:
The data type for exponent on the native is int8_t and from manage it is int (4bytes), so that is the most probable cause.

[newt nimble](https://github.com/apache/mynewt-nimble/blob/089f0d849a49a2a6c69f1e1f9103174b81af8b1a/apps/mesh_badge/src/gatt_svr.c#L41)
[Zehpyr Ble - nimble implementation](https://docs.zephyrproject.org/apidoc/latest/structbt__gatt__cpf.html)

closed 2022-06-08 by AdrianSoundy