nanoframework/Home #1074: Add StyleCop as a linter in managed code repositories

Ellerbach, 2022-06-08:
### Details about Problem

**nanoFramework area:** Repositories

**VS version:** All Visual Studio versions

**VS extension version:** All extensions

**Target:** N/A

**Firmware image version:** N/A

### Description

Core idea is to reduce PR commentsregarding code styles like brackets, intellisense comments missing, etc

- [ ] Add StyleCop as a linter in managed code repositories
- [ ] Find an automatic solution for the all IoT Device repository
- [ ] Setup proper rules in line with the .NET IoT repository

Ellerbach, 2022-06-08:
Some investigation already available: https://github.com/nanoframework/nanoFramework.IoT.Device/pull/423#discussion_r888064978

Ellerbach, 2022-06-08:
Getting to a point investigation seems to be done :-)
[image]

torbacz, 2022-06-08:
Looks very good, I assume that compilation will fail with this rules?

BTW If you need any help (testing/coding) feel free to reach me :)

Ellerbach, 2022-06-09:
> Looks very good, I assume that compilation will fail with this?

Yes, the idea is to treat linter issues as failure. So it has to be fixed before merging.

> BTW If you need any help (testing/coding) feel free to reach me :)

I take your words :-) So I need help here :-) Mainly because I won't have time and I'm a bit blocked ;-)
Here is how you can help me:

* Find a way to automate in the IoT Repository (https://github.com/nanoframework/nanoFramework.IoT.Device) for all the devices.
* Test and build few of them to understand if it's working, rules to adjust.

Here is the work I've been doing so far and how to use it:

* This needs to be added to all the packages.config: `<package id="StyleCop.MSBuild" version="6.2.0" targetFramework="netnano1.0" developmentDependency="true" />`
* I've added in the root a Directory.Build.props. This line does the trick to add the linter everywhere: `<Import Project="$(SolutionDir)\packages\StyleCop.MSBuild.6.2.0\build\StyleCop.MSBuild.targets" Condition="$(Configuration)=='Debug'" />`
* I'm now trying to add to each projects **without** having to copy/paste the Settings.StyleCop file in all directories using the Directory.Build.props. I've been failing here :-)
* Tip: to edit the config file if you need, there is an exe to edit the config when installing the nuget. Just use it like `.\StyleCop.SettingsEditor.exe pathto\Settings.StyleCop`

This branch contains this work: https://github.com/nanoframework/nanoFramework.IoT.Device/tree/add-linter
Files changed are just the package.config from the 4Relay, the Settings.StyleCop file in the StyleCop directory and the Directory.Build.props file.

So work left:

* Try to find a way to have the StyleCop file referenced everywhere without having to copy/paste it. If needed, we'll just copy/paste
* Find a way to adjust all needed package config in all the /devices directory including the samples (a bold find and replace should work)
* And of test this!

torbacz, 2022-06-09:
I will take a look during weekend.

torbacz, 2022-06-11:
@Ellerbach 
I've found something. We can execute custom msbuild command when running pipeline build. It should be pretty easy to implement - just few more arguments to pass.

Pros:
- Easy to implement across multiple projects (not only in this repo)
- No need to copy settings
- No need to modify any references
- Git submodule can be used for other projects, where msbuild option won't be enough 
- You can use different repository for settings and just pull it during build (one settings file across all projects). Also the stylecop msbuild can be referenced from that repository.

Cons:
- No errors during development
- I had to add /warnaserror flag to msbuild. It is ignoring ViolationsAsErrors=True from settings.stylecop file.

msbuild /p:CustomAfterMicrosoftCSharpTargets="C:\PrivateSources\nanoFramework.IoT.Device\devices\4Relay\packages\StyleCop.MSBuild.6.2.0\build\StyleCop.MSBuild.targets" /p:SourceAnalysisOverrideSettingsFile="C:\PrivateSources\nanoFramework.IoT.Device\StyleCop\Settings.stylecop" /warnaserror .\Tm1637.nfproj

Note: 
-  Make sure you use correct paths in msbuild (it's not throwing error when file does not exists...)
-  I removed Directory.Build.props file

Ellerbach, 2022-06-12:
Thanks a lot for the investigation. My main concern is clearly this one:

> * No errors during development

It's going to be very very very frustrating for developers not to see any error during the development. As I cannot test it right now, is it showing error when you build? If so, then it can be ok. all developers are building their projects to at least test them! And the linter should be available offline, for any developer, not just in the pipeline.

So if that can be done locally, then I'm all for it!

torbacz, 2022-06-12:
Yes, If you run it that way, it will show errors in console. So if you can modify the way that nanoframework is invoking msbuild for all projects - you are good to go. If not, let me know I'll try to find other way.

josesimoes, 2022-06-13:
We have targets specific to nanoFramework that are available to every build so we can introduce this as part of the regular build.

Now, we can't force everyone to use Stylecop rules, so this has to be made in such a way that it will only run if the stylecop is referenced. Otherwise it will silently ignore it.

Ellerbach, 2022-06-13:
> Now, we can't force everyone to use Stylecop rules, so this has to be made in such a way that it will only run if the stylecop is referenced.

We **can** actually force everyone to use a linter for everything related to the nanoFramework organization. We **cannot** force everyone to use it outside of our organization. 
Having a linter forced is what's done in the nf-interpreter side ;-) It just **must** be in the development chain, not on the pipeline only. Developer should have it since it creates a project.

So we should not add the linter to the nftarget by default because obviously, we cannot force outside of the org. So it should be per project. Like for the version.json and some other elements, we may have to copy the StyleCop file and add the additional nuget like for GitVersioning. That will allow to force to use for our nanoFramework organization but leave every developer free to not use it outside of the organization.

I hope it's more clear like that.

josesimoes, 2022-06-13:
Oooh! That's perfectly clear! And my thoughts too. Sorry if didn't manage to put it with enough clarity.  😅

Ellerbach, 2022-06-13:
> Oooh! That's perfectly clear! And my thoughts too. Sorry if didn't manage to put it with enough clarity. 😅

Perfect then :-) So seems like the build.prop with the nuget is a good way to go for the IoT repo (not needed for the others).
And copy the Settings.StyleCop inn each needed folder. And that's then a no brainer and simple.

Ellerbach, 2022-06-16:
> @Ellerbach Laurent Ellerbach FTE Anything I can help you with?

If you have time (which I currently don't have):

- Add back Directory.Build.props. This line does the trick to add the linter everywhere: <Import Project="$(SolutionDir)\packages\StyleCop.MSBuild.6.2.0\build\StyleCop.MSBuild.targets" Condition="$(Configuration)=='Debug'" />
- Copy the Settings.StyleCop file in all the devices/XXXX directories
- Add in all the nfproj projects the Settings.StyleCop file
- Add to all the packages.config: <package id="StyleCop.MSBuild" version="6.2.0" targetFramework="netnano1.0" developmentDependency="true" />

Not sure about the sample nfproj, we can skip for now.

And if you make a PR, make it draft :-) Because I guess there will be "few" things to fix :-)

More realistically, let's add the previous steps in like 5 devices and see how it behave

torbacz, 2022-06-29:
@Ellerbach 
Good news, I've manage to get it working with csproj. 
[image]

All steps commited to test repo: https://github.com/torbacz/testStyleCopMsBuild/commits/main

Next step: 
1. Follow the same steps on nfproj, make sure that it's working with nfproj project type (should, as you proven before)
2. Try to add Import/Error section to Directory.Build.props (based on https://github.com/torbacz/testStyleCopMsBuild/commit/fec7b925e94dffb59774658e3a5813848a899e31). 
3. If can't make it work through Directory.Build.props (I've noticed a problem, with current path in .props files and the stylecop target file is not imported) then extend the script to add sections.
4. Check custom settings, including StyleCopTreatErrorsAsWarnings in nfproj
5. Create PR :)

I will keep you posted.

josesimoes, 2022-06-29:
@torbacz horray!!! You're rocking with this. Thank you for this very much welcomed improvement. 💯 👏🏻

Ellerbach, 2022-06-30:
@torbacz HUGE thanks for finding all this and for the efforts! Can't wait to see this on our lovely Device.IoT repo :-) Thanks again!

torbacz, 2022-07-04:
Update:
1. I was able to make StyleCop work with nfproj
2. I wasn't able to make StyleCop work using Directory.Build.props, instead I've extend sync script
3. Currently the script is able to 
-  Sync settings from main folder to all devices (excluding samples for now, if we need samples also it's matter of removing one if from code)
- Sync StyleCopTreatErrorsAsWarnings setting (one critical bug here to fix)
- Create Import and Target sections in nfproj files (one improvement here todo, check all data in section and sync them if needed)
- Sync data in packages.config (based on consts from script)

Sync script still needs to be polished and tested. It may be run as a part of CI to make sure StyleCop is added for each new project. 

Current version of script: https://github.com/torbacz/nanoFramework.IoT.Device/blob/add-linter/StyleCop/syncSettingsOverEveryDevice.ps1

torbacz, 2022-07-05:
@josesimoes @Ellerbach 
Check this out and tell me if any adjustment are needed.
https://github.com/nanoframework/nanoFramework.IoT.Device/pull/484

With this settings both local build and AZDO build is failing due to StyleCop errors. Script is ready to be used for all projects also samples if needed.

josesimoes, 2022-08-03:
Now being tracked in #1101.

closed 2022-08-03 by josesimoes