nanoframework/Home #1080: GPIO debounce is not working for STM32

Scriptman, 2022-06-14:
### Library/API/IoT binding

System.Device.Gpio

### Visual Studio version

VS2022 (v17.1.1)

### Target name(s)

ST_NUCLEO64_F411RE_NF

### Firmware version

CLR Version: 1.8.0.98

### Device capabilities

Target capabilities:
  Has nanoBooter: YES
  nanoBooter: v21845.21845.21845.21845
  IFU capable: NO
  Has proprietary bootloader: NO

AppDomains:

Assemblies:
  Testing_Debounce, 1.0.0.0
  System.Threading, 1.0.4.3
  nanoFramework.Runtime.Events, 1.10.0.3
  System.Device.Model, 1.1.0.0
  mscorlib, 1.12.0.4
  UnitsNet.Length, 4.132.0.0
  Iot.Device.Hcsr04, 1.1.0.0
  System.Diagnostics.Stopwatch, 1.1.0.0
  System.Device.Adc, 1.0.2.4
  System.Device.Gpio, 1.0.4.3
  nanoFramework.Runtime.Native, 1.5.4.3
  System.IO.Ports, 1.0.7.6
  nanoFramework.System.Text, 1.1.3.3
  Iot.Device.Multiplexing, 1.1.0.0
  nanoFramework.System.Collections, 1.4.0.3
  System.Device.I2c, 1.0.3.3
  nanoFramework.Hardware.Stm32, 1.8.3.3
  System.Device.Spi, 1.2.1.12878
  System.IO.Streams, 1.0.0.2
  System.Device.Pwm, 1.0.1.3

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Stm32 v100.0.4.4, checksum 0x0874B6FE
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.0, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0xB6C4B3BD

### Description

When setting The DebounceTimeout for a Gpio Pin, the event callback is never called.

When disabling the DebounceTimeout, the event callback is working.

### How to reproduce

Use the sample code provided below.

### Expected behaviour

The event being called while DebounceTimeout set.

### Sample project or code

```
using System;
using System.Device.Gpio;
using System.Diagnostics;
using System.Threading;

namespace Testing_Debounce
{
    public class Program
    {
        public static void Main()
        {
            Debug.WriteLine("Starting program");

            GpioController gpioController = new();
            GpioPin gpioPin = gpioController.OpenPin(PinNumber('C', 11), PinMode.InputPullUp);

            gpioPin.DebounceTimeout = new TimeSpan(0, 0, 0, 100);
            gpioController.RegisterCallbackForPinValueChangedEvent(PinNumber('C', 11), PinEventTypes.Falling, PinStateChanged);

            Thread.Sleep(Timeout.Infinite);
        }

        public static void PinStateChanged(object sender, PinValueChangedEventArgs pinValueChangedEventArgs)
        {
            Debug.WriteLine("Pin state changed");
        }

        public static int PinNumber(char port, byte pin)
        {
            if (port < 'A' || port > 'J')
                throw new ArgumentException();

            return ((port - 'A') * 16) + pin;
        }
    }
}
```

networkfusion, 2022-06-15:
Just wondering if the solution is in this commit... https://github.com/nanoframework/nanoFramework.IoT.Device/commit/150c4d2ec3581f49b955eec0936452b83e76a1cd

josesimoes, 2022-06-16:
> Just wondering if the solution is in this commit... [nanoframework/nanoFramework.IoT.Device@150c4d2](https://github.com/nanoframework/nanoFramework.IoT.Device/commit/150c4d2ec3581f49b955eec0936452b83e76a1cd)

Nope. Debounce of GPIO pins it's carried in native code...

Scriptman, 2023-01-08:
Some time has passed, and I tried also tried setting the event via the GpioPin.ValueChanged, to no avail :(

`gpioPin.ValueChanged += gpioPin_ValueChanged`

josesimoes, 2024-12-26:
This should be fixed now. Please reopen if not.