nanoframework/Home #1081: Touch and Pin Deep Sleep Wakeup Doesn't Work on ESP32

rschiefer, 2022-06-15:
### Library/API/IoT binding

ESP32

### Visual Studio version

VS2022 (17.1.1)

### .NET nanoFramework extension version

2022.2.0.13

### Target name(s)

ESP32_REV0

### Firmware version

1.8.0.279

### Device capabilities

** .NET nanoFramework extension v2022.2.0.13 loaded **
GitHub repo: https://github.com/nanoframework/Home
Report issues: https://github.com/nanoframework/Home/issues
Browse samples: https://github.com/nanoframework/samples
Join our Discord community: https://discord.gg/gCyBu8T
Join our Hackster.io platform: https://www.hackster.io/nanoframework
Follow us on Twitter: https://twitter.com/nanoframework
Follow our YouTube channel: https://www.youtube.com/c/nanoFramework
Star our GitHub repos: https://github.com/nanoframework/Home
Add a short review or rate the VS extension: https://marketplace.visualstudio.com/items?itemName=nanoframework.nanoFramework-VS2022-Extension

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4.1
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Jun  3 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.8.0.279
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:
  ConnectedFeederTest, 1.0.0.0
  System.Device.I2c, 1.0.3.3
  nanoFramework.Runtime.Native, 1.5.4.3
  System.Buffers.Binary.BinaryPrimitives, 1.1.0.0
  Iot.Device.Vl53L0X, 1.1.0.0
  System.Diagnostics.Stopwatch, 1.1.0.0
  nanoFramework.Hardware.Esp32, 1.3.6.3
  System.IO.Streams, 1.0.0.2
  nanoFramework.System.Collections, 1.4.0.3
  mscorlib, 1.12.0.4
  System.IO.Ports, 1.0.7.1
  nanoFramework.Hardware.Esp32.Rmt, 1.2.3.5
  System.Device.Gpio, 1.0.4.3
  nanoFramework.Runtime.Events, 1.10.0.3
  System.Math, 1.4.4.3
  nanoFramework.System.Text, 1.1.3.3

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x0A915860
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.0, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0xB6C4B3BD
  System.Device.Wifi v100.0.6.4, checksum 0x1C1D3214
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

Deployment Map
Empty

### Description

Followed sample for deep sleep and wake.  Timer wake worked.  Touch and gpio pin wake did not.

### How to reproduce

Use deep sleep/wake sample

### Expected behaviour

ESP32 wakes up from deep sleep on touch or pin when following sample.

MichaelWheeley, 2022-06-15:
I'm assuming the code sample mentioned is this one? [Hardware.ESP32](https://github.com/nanoframework/Samples/tree/main/samples/Hardware.Esp32)

It is unfortunate in the sample code that many items are commented out. 
It is not clear what combination is required to create a clean wake on touch. Reading Arduino examples I think that Sleep.EnableWakeupByPin(WakeupGpioPin pin, int level) is a prior requirement for touch since the sensitivity is defined in the 'level' argument. Then the call to Sleep.EnableWakeupByTouchPad(). 

It is also not clear in the following code that it is an analog level and not a logic level. For logic levels it appears Sleep.EnableWakeupByMultiPins() is preferred.
`
                //    // Wakeup when Pin12 is high
                //    Sleep.EnableWakeupByPin(Sleep.WakeupGpioPin.Pin12, 1);
`

p.s. I'm writing these notes prior to my further testing, so please consider these my observations only.

p.p.s. I was relating this to what I read here [Arduino examples](https://randomnerdtutorials.com/esp32-deep-sleep-arduino-ide-wake-up-sources)

rschiefer, 2022-06-15:
Confirmed this configuration successfully enables touch, thanks!

            Sleep.EnableWakeupByMultiPins(Sleep.WakeupGpioPin.Pin4, Sleep.WakeupMode.AnyHigh);
            Sleep.EnableWakeupByTouchPad();
            Sleep.StartDeepSleep();

MichaelWheeley, 2022-06-15:
nice!

- Sample might need some updates to make it clearer, if I have time I can have a go
- I generated [pull](https://github.com/nanoframework/nanoFramework.Hardware.Esp32/pull/164) with changes to Sleep documentation to make it clearer

mrw

OaimIsGnay, 2022-11-24:
Which screen model can I use?

OaimIsGnay, 2022-11-24:
> Confirmed this configuration successfully enables touch, thanks!
> 
> ```
>         Sleep.EnableWakeupByMultiPins(Sleep.WakeupGpioPin.Pin4, Sleep.WakeupMode.AnyHigh);
>         Sleep.EnableWakeupByTouchPad();
>         Sleep.StartDeepSleep();
> ```

Which screen model can I use?

closed 2023-02-15 by josesimoes