nanoframework/Home #1082: When can I have a Bluetooth client?

bbhxwl, 2022-06-20:
### Description

When can I have a Bluetooth client?

### How to solve the problem

When can I have a Bluetooth client?

torbacz, 2022-06-20:
Isn't it already implemented?
https://github.com/nanoframework/Samples/tree/2cd29301b8648af5eadb31f9984f78892e7d4d0b/samples/Bluetooth

networkfusion, 2022-06-20:
It is indeed: https://github.com/nanoframework/nanoFramework.Device.Bluetooth

Please use Discord for further questions.

closed 2022-06-20 by networkfusion

bbhxwl, 2022-06-21:
> It is indeed: https://github.com/nanoframework/nanoFramework.Device.Bluetooth
> 
> Please use Discord for further questions.

It seems that there is no Bluetooth client yet. This is only the library of the Bluetooth server