nanoframework/Home #1087: Wifi Static IP assignment does not work

alberk8, 2022-06-28:
### Library/API/IoT binding

nanoFramework.System.Device.Wifi

### Visual Studio version

VS2022 v 17.2.5

### .NET nanoFramework extension version

2022.2.0.19

### Target name(s)

ESP32_REV0, ESP_PSRAM_REV0

### Firmware version

1.8.0.344, 1.8.0.291

### Device capabilities

System Information
HAL build info: nanoCLR running @ ESP32 built with ESP-IDF v4.4.1
  Target:   ESP32
  Platform: ESP32

Firmware build Info:
  Date:        Jun 28 2022
  Type:        MinSizeRel build, chip rev. >= 0, without support for PSRAM
  CLR Version: 1.8.0.379
  Compiler:    GNU ARM GCC v8.4.0

OEM Product codes (vendor, model, SKU): 0, 0, 0

Serial Numbers (module, system):
  00000000000000000000000000000000
  0000000000000000

Target capabilities:
  Has nanoBooter: NO
  IFU capable: NO
  Has proprietary bootloader: YES

AppDomains:

Assemblies:

Native Assemblies:
  mscorlib v100.5.0.17, checksum 0x004CF1CE
  nanoFramework.Runtime.Native v100.0.9.0, checksum 0x109F6F22
  nanoFramework.Hardware.Esp32 v100.0.7.3, checksum 0xBE7FF253
  nanoFramework.Hardware.Esp32.Rmt v100.0.3.0, checksum 0x0A915860
  nanoFramework.Device.OneWire v100.0.4.0, checksum 0xB95C43B4
  nanoFramework.Networking.Sntp v100.0.4.4, checksum 0xE2D9BDED
  nanoFramework.ResourceManager v100.0.0.1, checksum 0xDCD7DF4D
  nanoFramework.System.Collections v100.0.1.0, checksum 0x2DC2B090
  nanoFramework.System.Text v100.0.0.1, checksum 0x8E6EB73D
  nanoFramework.Runtime.Events v100.0.8.0, checksum 0x0EAB00C9
  EventSink v1.0.0.0, checksum 0xF32F4C3E
  System.IO.FileSystem v1.0.0.0, checksum 0x3AB74021
  System.Math v100.0.5.4, checksum 0x46092CB1
  System.Net v100.1.5.0, checksum 0x5BAB8CB3
  System.Device.Adc v100.0.0.0, checksum 0xE5B80F0B
  System.Device.Dac v100.0.0.6, checksum 0x02B3E860
  System.Device.Gpio v100.1.0.4, checksum 0xB6D0ACC1
  System.Device.I2c v100.0.0.1, checksum 0xFA806D33
  System.Device.Pwm v100.1.0.4, checksum 0xABF532C3
  System.IO.Ports v100.1.6.0, checksum 0xB798CE30
  System.Device.Spi v100.1.2.0, checksum 0xB6C4B3BD
  System.Device.Wifi v100.0.6.4, checksum 0x1C1D3214
  Windows.Storage v100.0.2.0, checksum 0x954A4192

++++++++++++++++++++++++++++++++
++        Memory Map          ++
++++++++++++++++++++++++++++++++
  Type     Start       Size
++++++++++++++++++++++++++++++++
  RAM   0x3ffe49ac  0x0001b000
  FLASH 0x00000000  0x00400000

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
++                   Flash Sector Map                        ++
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  Region     Start      Blocks   Bytes/Block    Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
      0    0x00010000       1      0x1A0000     nanoCLR
      1    0x001B0000       1      0x1F0000     Deployment
      2    0x003C0000       1      0x040000     Configuration

+++++++++++++++++++++++++++++++++++++++++++++++++++
++              Storage Usage Map                ++
+++++++++++++++++++++++++++++++++++++++++++++++++++
  Start        Size (kB)           Usage
+++++++++++++++++++++++++++++++++++++++++++++++++++
  0x003C0000    0x040000 (256kB)    Configuration
  0x00010000    0x1A0000 (1664kB)   nanoCLR
  0x001B0000    0x1F0000 (1984kB)   Deployment

### Description

Assigning Static IP and connecting to an AP.  The connection is successful and the Console message show the assigned IP of the ESP32 but viewing the AP GUI will reveal that an DHCP IP is assigned instead. 

### How to reproduce

1. Reset AP DHCP service
2. Get the example code from [Static IP Code](https://github.com/alberk8/nanoFramework-StaticIP)
3. Change the SSID, Password, IP, GW and DNS addresses.
4. Compile and Run code
5. View the messages
6. Check the DHCP on the AP

### Expected behaviour

Static IP is used instead of DHCP.

lsd-techno, 2022-06-28:
Seems You are first person to use static IP in Wifi client mode.
after `WifiNetworkHelper.ConnectFixAddress(......)`
`ni.IPv4Address[0]` returns/holds desired IP address, but actual IP address assigned by router's DHCP.

tested on target: ESP_PSRAM_REV0 FW: 1.8.0.291

cbagpipe, 2022-07-12:
caught same bug ESP32_REV0 FW: 1.8.0.431

josesimoes, 2022-08-23:
[Question from Stackoverflow](https://stackoverflow.com/questions/73403557/nanoframework-esp32-cant-no-set-fix-ip-address)

Ellerbach, 2022-08-23:
So I've been doing some investigation and flashed an ESP32 (REV0) with the latest firmware.

[image]

The configuration is properly stored:

[image]

[image]

[image]

It does connect properly, I tried different scenarios:

* First time connection with a fully mass erased flash
* Second and third connection after a successful connection

I've been trying as well non proper connection, in those cases, as expected all goes wrong, you can't connect.

I've been trying the sample https://github.com/nanoframework/Samples/blob/main/samples/HTTP/HttpWebRequest/Program.cs and it works just fine.

alberk8, 2022-08-24:
The issue is with setting the static IP on the ESP32, where the router is not having a DHCP server running.  I will not connect in my case.  When you have a DHCP server running on the Router/Gateway the ESP32 will be provisioned with it.  So the static IP that you have provisioned in ESP32 is not used at all, the static IP will show in ESP32.   

One way to test it is to ping the ESP32 static IP and I am sure you will get time out and also check with DHCP server it will show that it has provisioned with a Dynamic IP to ESP32.

cbagpipe, 2022-10-06:
I searched through espressif documentation, and found, that

[image]

`esp_netif_set_ip_info` used only one time here — [https://github.com/nanoframework/nf-interpreter/blob/31224a304d1fe887e4f700c07287f8af524dc803/targets/ESP32/_Network/NF_ESP32_Wireless.cpp#L389](https://github.com/nanoframework/nf-interpreter/blob/31224a304d1fe887e4f700c07287f8af524dc803/targets/ESP32/_Network/NF_ESP32_Wireless.cpp#L389)

So I searched, how to disable DHCP client, and found

[image]

Then searched through `nf-interpreter` repo and found nothing

[image]

So, static IP not implemented properly, I think.

Hope this information helps someone, because I'm not c++ programmer and can't handle with it myself

cbagpipe, 2023-02-14:
```csharp
using nanoFramework.Networking;
using System.Threading;

    public class Program
    {
        public static void Main()
        {
            // connecting to wifi ap
            bool connected;
            do
            {
                WifiNetworkHelper.SetupNetworkHelper("SSID", "PWD");
                WifiNetworkHelper.NetworkReady.WaitOne();
                connected = WifiNetworkHelper.ConnectFixAddress("SSID", "PWD", new IPConfiguration("192.168.45.115", "255.255.255.0", "192.168.45.1"));
            } while (!connected);

            Thread.Sleep(Timeout.Infinite);
        }
    }
```

With this simple code and DHCP server in local network  I have unpredictable result. Sometimes ESP32 gets IP from DHCP, sometimes static IP. I have to ping static IP from my PC and reboot board several time, to catch a moment, when static IP assigned.

Ellerbach, 2023-02-16:
@cbagpipe what you are doing is not really a good way. Why would you first connect with DHCP and then with a static address?

You should just use `connected = WifiNetworkHelper.ConnectFixAddress("SSID", "PWD", new IPConfiguration("192.168.45.115", "255.255.255.0", "192.168.45.1"));` and not the 2 previous lines. Than it will work as expected.

cbagpipe, 2023-02-16:
@Ellerbach ok... I thought `WifiNetworkHelper.SetupNetworkHelper` does some initialization without connecting, because we have `WifiNetworkHelper.ConnectDhcp`.

Ellerbach, 2023-02-17:
> I thought `WifiNetworkHelper.SetupNetworkHelper` does some initialization without connecting, because we have `WifiNetworkHelper.ConnectDhcp`.

It does actually connect and setup everything :-) See: https://github.com/nanoframework/System.Device.Wifi/blob/6767df36c7dfd44caf651f21b52f616d57b3f33c/System.Device.Wifi/NetworkHelper/WifiNetworkHelper.cs#L65
To me, the intellisense comment seems very clear as well.

cbagpipe, 2023-02-22:
> @cbagpipe what you are doing is not really a good way. Why would you first connect with DHCP and then with a static address?
>
> You should just use `connected = [...]

After doing that my ESP32 not getting static IP anymore. Only DHCP one.

Ellerbach, 2023-03-01:
@cbagpipe thanks for the investigation. So definitely something to adjust on the native side.

Scriptman, 2023-03-16:
Having the same issue since januari with stm32f411re. Also static ip being ignored. So didnt investigate further and left as is.

networkfusion, 2024-05-05:
@alberk8 , Please can you see if this is now working.

alberk8, 2024-05-06:
> @alberk8 , Please can you see if this is now working.

Tested with latest firmware and still not working with static IP with WiFi.   It is still getting DHCP IP from my router.

AdrianSoundy, 2024-06-10:
This has been an issue since the IDF changed from TCPIP adapters to the ESP_NETIF interface.
The problem is the ESP_NETIF starts the DHCP when the WIFI connects overriding the static IP settings.

Some work needs to be done to make sure all areas are compatible with ESP_NETIF working.
- When configuring Wifi STA on start make sure DHCP is disabled if config is for static
- When config is changed from managed app make sure ESP_NETIF is also updated.

belem2050, 2025-01-22:
By enabling staticipV4, it worked for me:

```c#
IPConfiguration IPConfiguration = new IPConfiguration("192.168.31.147", "255.255.255.0", "192.168.31.1", new string[] {"8.8.8.8"});

// 0 was the index of the interface I wanted to connect to

 NetworkInterface.GetAllNetworkInterfaces()[0].EnableStaticIPv4(IPConfiguration.IPAddress, IPConfiguration.IPSubnetMask, IPConfiguration.IPGatewayAddress);
success = WifiNetworkHelper.ConnectFixAddress(MySsid, MyPassword, IPConfiguration, token: cs.Token);
```

AdrianSoundy, 2025-03-14:
> [@AdrianSoundy](https://github.com/AdrianSoundy) is this completed now?

No it wasn't completed. The changes have been made but the testing wasn't completed. I'll raise a PR for changes and see if we can get this tested.  @alberk8 Are you able to test this ?

alberk8, 2025-03-14:
> > [@AdrianSoundy](https://github.com/AdrianSoundy) is this completed now?
> 
> No it wasn't completed. The changes have been made but the testing wasn't completed. I'll raise a PR for changes and see if we can get this tested. [@alberk8](https://github.com/alberk8) Are you able to test this ?

I will try to test this over the week end.