nanoframework/Home #1089: Json library does not deserialize Version type

axa88, 2022-07-02:
### Library/API/IoT binding

nanoFramework.Json

### Visual Studio version

VS2019

### .NET nanoFramework extension version

v2019.11.0.23

### Target name(s)

ESP32_BLE_REV0

### Description

I am able to serialize but NOT deserializing type `System.Version`

I didnt notice a [unit test](https://github.com/nanoframework/nanoFramework.Json/blob/main/nanoFramework.Json.Test/JsonUnitTests.cs) for the type
So im left wondering if perhaps there is or not full support for this type.

Basically I have the string including this 'Version' object:
    ```
"Version": {
        "Major": 0,
        "Build": 0,
        "Revision": 1,
        "Minor": 0
    }
```

but it will not deserialize into the type that includes a Version type

```
internal class ConfigData
{
    public Version Version { get; set; }
    public byte[] PassKey { get; set; }
    public bool Ignition { get; set; }
    public Mode IgnitionMode { get; set; }
    public byte[] IgnitionCombo { get; set; }
    public ConfigurationBase IndicatorConfiguration { get; set; }
}
```

Where everything else deserializes fine.

I pushed a condensed [test app](https://github.com/axa88/NFTestApp) for brevity, showing the problem at line 116.
There still a bit of code ceremony, but i didnt want to change the fundamentals of the code just incase i was doing something incorrectly.  Regardless the problem persists.

### How to reproduce

Attempt to deserialize a `System.Version` type 
Alternatively load and observe the exception thrown in the example app:
https://github.com/axa88/NFTestApp

### Expected behaviour

Its expected not only to serialize, but deserialize a `System.Version` object

### Sample project or code

https://github.com/axa88/NFTestApp

copied below:

```
using System;
using System.Collections;
using System.Diagnostics;
using System.IO;
using System.Text;
using System.Threading;

using nanoFramework.Json;

namespace NFTestApp
{
	public class Program
	{
		private const string Path = "I:\\ConfigData.json";
		private static ConfigData _configData = new();
		private static IDictionary _data = new Hashtable();

		public static void Main()
		{
			var configRead = ReadConfig();
			var version = new Version(0, 0, 0, 1);
			var equals = Equals((Version)_data[NVElement.Version], version);
			if (!configRead || !equals)
			{
				#if DEBUG
				Debug.WriteLine("NVS Re/Uninitialized");
				#else
				#endif

				const byte mx = byte.MaxValue;
				_data = new Hashtable
				{
					{ NVElement.Version, version },
					{ NVElement.PassKey, new byte[] { 0, 0, 0, 0 } },
					{ NVElement.Ignition, false },
					{ NVElement.IgnitionMode, Mode.Invalid },
					{ NVElement.IgnitionCombo, new[] { mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx, mx } },
					{ NVElement.Indicators, new ConfigurationBase() }
				};

				Thread.Sleep(Timeout.Infinite);
			}
		}

		internal enum NVElement
		{
			Version,
			PassKey,
			Ignition,
			IgnitionMode,
			IgnitionCombo,
			Indicators,
			IndicatorButtonMode,
			IndicatorRate,
			IndicatorOffMode,
			IndicatorOffTime,
			IndicatorOffAngle
		}

		public enum ButtonMode
		{
			Momentary,
			Latched,
			Invalid = byte.MaxValue
		}

		public enum Rate
		{
			Slow,
			Moderate,
			Fast,
			Illegal,
			Count,
			Invalid = byte.MaxValue
		}

		public enum OffMode
		{
			Manual,
			Timer,
			Compass,
			TimerAndCompass = Timer | Compass,
			Count,
			Invalid = byte.MaxValue
		}

		internal enum Mode
		{
			Key,                            // no bt, no controls, ignition on with power.
			Controls,                       // no bt, ignition with control dance, sequence needs to be set
			Bt,                             // ignition on with bt proximity
			BtAndControls = Controls | Bt,  // bt and controls.
			Invalid = byte.MaxValue         // max value for bad read
		}

		private static bool ReadConfig()
		{
			FileStream fileStream = null;

			if (File.Exists(Path))
			{
				try
				{
					fileStream = new(Path, FileMode.Open, FileAccess.Read);
					var buffer = new byte[fileStream.Length];
					if (fileStream.Read(buffer, 0, (int)fileStream.Length) > 0)
					{
						var s = Encoding.UTF8.GetString(buffer, 0, buffer.Length);

						#if DEBUG
						Debug.WriteLine(s);
						#else
						#endif

						var o = JsonConvert.DeserializeObject(s, typeof(ConfigData)); // TODO throws
						_configData = (ConfigData)o;

						_data[NVElement.Version] = _configData.Version;
						_data[NVElement.PassKey] = _configData.PassKey;
						_data[NVElement.Ignition] = _configData.Ignition;
						_data[NVElement.IgnitionMode] = _configData.IgnitionMode;
						_data[NVElement.IgnitionCombo] = _configData.IgnitionCombo;
						_data[NVElement.Indicators] = _configData.IndicatorConfiguration;

						return true;
					}
				}
				catch (Exception e)
				{
					Debug.WriteLine(e.Message);
					return false;
				}
				finally
				{
					fileStream?.Dispose();
				}
			}

			return false;
		}

		public class ConfigurationBase
		{
			// for deserialization
			public ConfigurationBase()
			{
			}

			protected ConfigurationBase(ConfigurationBase @base)
			{
				ButtonMode = @base.ButtonMode;
				Rate = @base.Rate;
				OffMode = @base.OffMode;
				OffDelay = @base.OffDelay;
				OffAngle = @base.OffAngle;
			}

			public ButtonMode ButtonMode { get; set; } = ButtonMode.Invalid;
			public Rate Rate { get; set; } = Rate.Invalid;
			public OffMode OffMode { get; set; } = OffMode.Invalid;
			public byte OffDelay { get; set; }
			public byte OffAngle { get; set; }

			#region Overrides of Object
			public override bool Equals(object obj) => obj is ConfigurationBase cb && ButtonMode == cb.ButtonMode && Rate == cb.Rate && OffMode == cb.OffMode && OffDelay == cb.OffDelay && OffAngle == cb.OffAngle;
			public override int GetHashCode() => (byte)ButtonMode ^ (byte)Rate ^ (byte)OffMode ^ OffDelay ^ OffAngle;
			#endregion
		}

		internal class ConfigData
		{
			public Version Version { get; set; }
			public byte[] PassKey { get; set; }
			public bool Ignition { get; set; }
			public Mode IgnitionMode { get; set; }
			public byte[] IgnitionCombo { get; set; }
			public ConfigurationBase IndicatorConfiguration { get; set; }
		}
	}
}
```

Ellerbach, 2022-07-03:
The way the Json lib is working is with default constructor and then setting the properties.
Version has read only properties (see https://docs.microsoft.com/en-us/dotnet/api/system.version?view=net-6.0) so you can't set the properties once it's been created. So it's the reason why it's not working.
You need an intermediate step or have your own Version class.

axa88, 2022-07-03:
@Ellerbach Thanks for the response.
 I do understand Json deserialization requires a default constructor in this way.
So I need to push back on this answer as if you notice the issue reported is merely that the `Version` property of `ConfigData` is the one not being deserialized. I expect then it uses the default constructor and should then set deserilzed data directly.  It is not then written over afterward by any over overloaded constructor.

`var o = JsonConvert.DeserializeObject(s, typeof(ConfigData));`

```
internal class ConfigData
{
	public Version Version { get; set; }
	public byte[] PassKey { get; set; }
	public bool Ignition { get; set; }
	public Mode IgnitionMode { get; set; }
	public byte[] IgnitionCombo { get; set; }
	public ConfigurationBase IndicatorConfiguration { get; set; }
}
```
		

Im sorry I should have simplified the example but was unsue of the cause i didnt want to modify from what wasnt working.
Ill simplify as it will help isolate the issue.

Ellerbach, 2022-07-04:
> But concider merely changing the type from `Version` to say `String` solves the issue.

Yes, as I wrote, the deserializer call the Version default constructor, then it sets the properties. But there is no property to set because they only have the getter, not the setter. If you have a string, then the string with a get and set will be deserialized properly.

In other words, let's assume you create your own Version class called Version2:

```csahrp
public class Version2
{
  public int Major { get; set; }
  public int Minor { get; set; }
}
```

This one will work (add the missing ones). But the Version from the framework has a different definition and simplified looks like:

public class Version
{
  public int Major { get; }
  public int Minor { get; }
}
```

And this one won't deserialize properly because there is no `set;` so there is no way to populate properly the property.
Does it makes more sense?

axa88, 2022-07-05:
@Ellerbach Ok i see now. What you are saying is that System.Version has no default constructor for nf like the full blown .net does:

```
        public Version()
        {
            _Major = 0;
            _Minor = 0;
        }
```

Seems simple enough, can this not be added to nf?

josesimoes, 2022-07-05:
@axa88 nope! Version properties **DO NOT** have a getter, therefore the deserializer **CAN NOT** populate them when trying to construct the object. Please take a look at the code [here](https://github.com/nanoframework/CoreLibrary/blob/main/nanoFramework.CoreLibrary/System/Version.cs)

torbacz, 2022-07-05:
@josesimoes 
I guess you meant setter not getter :)

Ellerbach, 2022-07-05:
> I guess you meant setter not getter :)

Correct :-) Setter :-)

If the function does not exists, there is no way to invent it. And we **cannot** use any of the existing constructor to pass the parameters as there is no way to know the order. Can't test right now but it's most likely something that will not work with the normal .NET.

axa88, 2022-07-06:
> Can't test right now but it's most likely something that will not work with the normal .NET.

And you would be right...   Realizing Ive used Json.Net exclusivly for years, would have sworn `Version` fully supported.
[So its a known limitation](https://stackoverflow.com/a/13177340/5122089):