nanoframework/Home #1091: JSON serializer produces invalid output for multiline strings

bauermaximilian, 2022-07-08:
### Library/API/IoT binding

nanoFramework.Json

### Description

Serializing strings containing newline characters will result in an invalid JSON output.
The generated JSON object string will actually have a line break inside the string, rendering the JSON string invalid.

### How to reproduce

Serializing a simple string with a newline character, for example ``multiline\nstring`` with ``nanoFramework.Json.JsonConvert.SerializeObject`` will create the following (invalid) JSON string:
```
["multiline
string"]
```

### Expected behaviour

The console output from the code example should be `["multiline\nstring"]`

### Sample project or code

```
string json = nanoFramework.Json.JsonConvert.SerializeObject("multiline\nstring");
Console.WriteLine(json);
```

### Aditional information

Similar behavior was observed with ``\r`` - an eventual bugfix should therefore also include other escaped control characters. Attempts to escape the backslash so the string would read ``multiline\\nstring`` didn't work either.

bauermaximilian, 2022-07-14:
Heya @torbacz,
thanks for the quick response and your PR - I checked it out, and this whole matter is a bit confusing, so I'll try to explain: A C# string like ``string test = "test\ntext"`` should actually be serialized into the JSON string ``"test\ntext"``, which is the equivalent of the C# string ``string jsonTest = "test\\ntext"``. So the console output of the generated JSON string having two lines is the actual bug. Maybe [this](https://stackoverflow.com/a/2392888/13223033) StackOverflow answer does a better job in explaining this problem.

I created a small test solution with two projects, which basically test (de-)serializing the same test string - one test uses the nanoFramework JSON nuget, the other one the "native" serializer from ``System.Text.Json``. While both tests pass, they produce different JSON output, as seen in the following screenshot:
[image]

This screenshot also exposes two other issues - the debug output of the nanoFramework test is truncated after a newline, and the JSON deserializer accepts a JSON string with an "actual" newline in it. Trying that with an online JSON validator will yield following error:
[image]
I am not sure if I should open another issue for the latter, as it's debatable whether this is a bug or a feature 😅

But regarding the current issue - I hope this explanation somewhat makes sense and doesn't add more confusion. I attached the test solution from the screenshot above: [JsonTest.zip](https://github.com/nanoframework/Home/files/9112851/JsonTest.zip). Let me know if I can do anything else to help!

josesimoes, 2022-07-14:
The issue with VS output could be related with the way we're sending the strings over to VS output pane... I don't think we have any extra processing for "escaping" these in our code. But we should have.

torbacz, 2022-07-15:
@bauermaximilian Now it's clear, but i see 2 issue there:
1. Json data for simple string is invalid 
2. Console.Writeline is creating new lines for "/n"

I've adjusted my tests and fixed issue, now JSON created from string should be correct:  https://github.com/nanoframework/nanoFramework.Json/pull/226

bauermaximilian, 2022-07-16:
@torbacz Okay, I checked it out - the removal of the braces (so that a string is serialized into "some string" and not ["some string"]) is good, thanks for that! 
The second issue issue can be resolved by serializing "multiline\nstring" into "multiline\\nstring". Then, the console will output "multiline\nstring", and that's what any JSON parser will accept as well. Or, to put it into other words, to generate valid multiline JSON strings, any newline in the source C# string needs to be replaced with a "\\n" in the target JSON string. Without that change, serializing any object with any string that contains any "real" newline will be invalid.
I originally came across this issue while making a REST API, and the JSON coming from my ESP32 device couldn't be parsed by the browser - so I am very sure that it's not an issue with the Console, but the Console.WriteLine only "exposes" that underlying bug.
In any case, thanks for your patience here!

bauermaximilian, 2022-07-18:
> But one good news I think that now I understand issue :D
Glad to hear - I mean, it is rather confusing, and I probably didn't do a good job in explaining it unambiguously :D
> And one more problem, This week I won't be able to continue my work on issue - I'm pretty busy.
No problem, thank you for investing time into this! When you find the time to work on it a bit more, I'll be happy to discuss this further and/or do a code review!

torbacz, 2022-07-31:
@bauermaximilian 
I've uploaded new code and I think it's final version, but not yet tested with real API. Would be nice if you could test it :)

bauermaximilian, 2022-08-03:
Apologies for the delay, but I finally got to test it - and what can I say...
[image]
I attached the little test project I wrote for this [here](https://github.com/nanoframework/Home/files/9254576/JsonTest.zip), if you want to give it a go as well.
But besides that, thanks a lot for your good work - I just approved your changes, let me know if you need anything else! 👌

torbacz, 2022-08-03:
@bauermaximilian 
Glad it works :)

PR is ready to review.

closed 2022-08-04 by josesimoes