nanoframework/Home #1098: [DependencyUpdater] Problem with custom NuGet repositories

torbacz, 2022-07-19:
### Tool

nanodu

### Description

Checking updates for customPackage.2.1.2
ERROR: nuget CLI exited with code 1

Package 'customPackage' is not found in the following primary source(s): 'https://pkgs.dev.azure.com/nanoframework/feed/_packaging/sandbox/nuget/v3/index.json'. Please verify all your online package sources are available (OR) package id, version are specified correctly.

The correct nuget source is present in nuget.config file, which seems to be ignored by nanodu.

josesimoes, 2022-07-19:
Please check what is **customPackage**.2.1.2 because we don't have such package on our feed...

josesimoes, 2022-07-19:
If you have/place a nuget.config at nfproj folder (or sln) folder, nuget will take the feeds that are registered there.
As an alternative you can add your feeds to global nuget config on the machine running the action.

torbacz, 2022-07-20:
Well, nuget.config exists in the same folder as sln and the visual studio can download packages correctly. I think the nanodu should behave in the same way, so if there is nuget.config file it should be passed into nuget command. 

I already got fix for that and I'll submit PR shortly.

torbacz, 2022-07-21:
Let me clarify this issue. 
So first, we've got our project which uses nanoFramework nugets and also some other nugets targeting nanoFramework, but in our custom nuget repository. There is a NuGet.config file with all required sources and the "sln" is building fine with VisualStudio.
After running nanodu on our repository it fails to resolve packages from our custom repository when **updating**, despite specifying the nuget sources in nuget.config file.

There were two issues:
1. Flag useNugetConfig when updating packages was set to false for update command
2. Specifying "-Source" explicitly in nuget arguments results in ignoring nuget.config file

I've also included code to check if we are using nuget.config file and if not I'm appending 2 default nuget sources. Done that to keep the compatibility with repositories where there is no nuget.config file.

And one more fix, git global settings are set only if we are **not** in debug mode. We don't want to change developers global git config file, when debugging tool.

@josesimoes Again, If you could check&merge as soon as possible. We are struggling with using nanoFramework in our project due to many changes in nanoFramework packages and it's hard to keep it up to date with manual work.

josesimoes, 2022-07-22:
@torbacz understood on the above. Request for future interactions: please add comments and reviews about what is going on on the respective PR. It gets easier to read and contextualize. 
It will still be linked to the Issue that originated the PR as the PR is linked to the Issue. 

(hope I'm making sense... 😉 )

josesimoes, 2022-07-22:
> And one more fix, git global settings are set only if we are **not** in debug mode. We don't want to change developers global git config file, when debugging tool.

Yes! I overlooked that on my quick fix to the fix.

> @josesimoes Again, If you could check&merge as soon as possible. We are struggling with using nanoFramework in our project due to many changes in nanoFramework packages and it's hard to keep it up to date with manual work.

Oh yes I hear you! That was the main driver to create this tool! 😅

closed 2022-07-22 by josesimoes

torbacz, 2022-07-22:
@josesimoes 
Thanks!

I can confirm that tool works with 3rd party repositories.

GitHub action:
[image]

[image]

Created PR:
[image]

josesimoes, 2022-07-22:
All right!! Thanks for the positive feedback. And more importantly, for improving it. 😉 👍