nanoframework/Home #1101: Apply StyleCop to IoT bindings and core libraries

josesimoes, 2022-07-22:
### Details about Problem

We now have a usable pattern and rules for using StyleCop in nf projects.
Need to add it to IoT bindings and the class libaries.

### Description

This task it's a bit tedious, granted, but has to be done. 
It's not complex, just a matter of adding a file and a couple of changes to the projects and then fixing the warnings from StyleCop.

Please check [this PR](https://github.com/nanoframework/nanoFramework.IoT.Device/pull/506) for reference.

**nanoFramework area:** ( Visual Studio extension | Other tools )

torbacz, 2022-07-26:
Info:
There is a script for syncing all projects in path: /StyleCop/syncSettingsOverEveryDevice.ps1
Please use it, if you want apply stylecop to project.

networkfusion, 2022-07-27:
For the core libs, More consise documentation and perhaps sample PR needed (since the one provided in the description is too broad to be of help). Although a great start is: https://github.com/nanoframework/nanoFramework.IoT.Device/tree/develop/StyleCop .

rkalwak, 2022-07-28:
I applied StyleCop on library I am working and and have a problem with below line, is this too new language feature usage that StyleCop complains with SA0102 : CSharp.CsParser : A syntax error has been discovered in file ...?

`if (value.DegreesCelsius is < -55 or > 125)`

torbacz, 2022-07-28:
@rkalwak 
Unfortunately it does not support some C# features and the analysis stops when meets this kind of code. As for workaround I suggest to replace
`if (value.DegreesCelsius is < -55 or > 125)`
with
`if (value.DegreesCelsius < -55 ||  value.DegreesCelsius> 125)`

PS. You can extract -55 and 125 to consts to keep code cleaner :)

Ellerbach, 2022-07-29:
> if (value.DegreesCelsius is < -55 or > 125)

Yes, the .NET Framework 4.x StyleCop that we are using here is not perfect for this. We will move to SDK base and we then *should* be able to use the modern one for .NET 5+ and .NET Core 3+ which is handling those elements properly. I'll try to check what we can do.
You can as well release the rule with a pragma for a block of code. Good temporary workaround as well.

torbacz, 2022-07-31:
I think the best approach to work on style cop fixes is:
1. Replace in syncSettingsOverEveryDevice.ps1 script
`$projectWhiteList = ("AD5328", "4Relay", "Ads1115", "Adxl345", "Adxl357", "Ags01db", "Ahtxx")`
with
`$projectWhiteList `
2. Run script.
3. Check git status for changed files.
4. Discard all changes from project you don't want to fix.
5. Fix projects one by one and push only one to repository (PR should be opened as draft to check if build is passing after applying fixes)

josesimoes, 2022-10-12:
Still several bindings to be worked out!

torbacz, 2022-11-13:
Few more to go.

Ellerbach, 2026-09-28:
I'll close this one as I feel we are in a great shape in the IoT repo. And all new devices are not enrolled to style cop. Thanks again @torbacz for the heavy lifting!

closed 2026-09-28 by Ellerbach