nanoframework/Home #1104: Bmm150 - tests not compiling.

torbacz, 2022-07-31:
### Library/API/IoT binding

Bmm150

### Visual Studio version

VS2022

### Description

Tests are not compiling. Found during style cop fixes. 
Also, are those tests run in pipeline? Maybe they should be?

### Screenshots

[image]

josesimoes, 2022-08-02:
I can confirm that Unit Tests aren't running in the pipeline. That's because the builds in IoT repo are handled substantially different than in the other repos: from the command line. But agreed that that should be improved.

josesimoes, 2022-08-02:
On the above: that's most likely some tweak on the unit test API that was introduced after the Bmm150 Unit Tests were coded...

closed 2024-12-26 by josesimoes