nanoframework/Home #1106: VS COM Exclusion does not work as expected.

networkfusion, 2022-08-02:
### Tool

Visual Studio extension

### Description

When using a USB (FTDI) device that has highly specific serial commands, having the nF VS extension can interfere with it and make communications impossible,

It seems like the "exclusion list" is ignored (see screenshot) and underlying cause is due to https://github.com/nanoframework/nf-debugger/blob/b272fe68c3dd3536115641b0a27ddf8b2982a51f/nanoFramework.Tools.DebugLibrary.Shared/PortSerial/PortSerialManager.cs#L190 (or something calling it without checking first).

### How to reproduce

Add a device to the exclusion list e.g. `COM15`
Plug it in
Unplug it
Plug it in
check output window for messages like`[NanoDevices: checking device  COM15 *** 2nd attempt ***]`

### Expected behaviour

The VS extension output window shows that the "serial port" was ignored.

### Screenshots

[image]

### Aditional context

VS2022 - 17.2.6

josesimoes, 2024-10-28:
@networkfusion I've been using this and even tested it with the last rework on the debugger library. Didn't notice anything wrong.
Could it be that this got fixed since the last time you've tested it?

networkfusion, 2024-10-28:
Very likely as it was 2 years ago. I will close for now.

closed 2024-10-28 by networkfusion